sparkfun/SparkFun_Ublox_Arduino_Library #5: I2C_BUFFER_LENGTH warning

trycoon, 2019-03-25:
Hi!

I get this warning when running the latest SparkFun Ublox Arduino Library in Platform.io on a ESP32:

`In file included from src/gps.h:5:0,
from src/gps.cpp:2:
.piolibdeps/SparkFun Ublox Arduino Library_ID5746/src/SparkFun_Ublox_Arduino_Library.h:71:0: warning: "I2C_BUFFER_LENGTH" redefined
#define I2C_BUFFER_LENGTH 32
^
In file included from .piolibdeps/SparkFun Ublox Arduino Library_ID5746/src/SparkFun_Ublox_Arduino_Library.h:38:0,
from src/gps.h:5,
from src/gps.cpp:2:
/home/trycoon/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:34:0: note: this is the location of the previous definition
#define I2C_BUFFER_LENGTH 128`

Would it be possible to set this value ONLY when it has not already been set? 
I2C_BUFFER_LENGTH 32

Thanks!

nseidle, 2019-03-25:
Yep. Probably all that is needed is to change this [line](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/master/src/SparkFun_Ublox_Arduino_Library.h#L68) to an #ifndef. If you can try it and do a PR I would be very grateful (I don't often do testing on ESP32).

trycoon, 2019-04-07:
@nseidle Sorry for the delay, but now there is a pull-request available: https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/pull/7

closed 2019-04-17 by nseidle