sparkfun/SparkFun_Ublox_Arduino_Library #8: no valset support

nathanWagenbach, 2019-04-13:
I have been trying to implement valset myself in this library, but have been unsuccessful so far. 
Has there been any progress made so far on getting valset working?

nseidle, 2019-04-17:
I haven't had the need for valset yet (all though I know I'll need it someday). What command or setting are you attempting that the current lib doesn't cover? It may help me troubleshoot.

nathanWagenbach, 2019-04-17:
I was attempting to enable high precision NMEA output

nseidle, 2019-04-28:
Unfortunately there's nearly 40 pages of various keyIDs and I can't figure out which one your poking at. Mind giving me some more information about which key you're trying to set? (Please provide as much info as possible in your issues. Makes it easier for me to replicate.)

Good news: I just committed an example demoing getVal of the I2C address. Once I know what key you're going after I can better write setVal.

nseidle, 2019-04-28:
Think I found it: CFG-NMEA-HIGHPREC 0x10930006?

nseidle, 2019-04-28:
While a key can be generated from a group and ID, I can't figure out Ublox's classifications. Let me know if you find a good source of Groups and IDs from Ublox. If we can't pass group/ID/size then we'll have to pass hard coded keys which is not ideal IMO.

nseidle, 2019-04-28:
Ok - I added a setVal() example (see ZED-F9P dir for Example7_setVal) that demonstrates how to set the measurement rate, high precision NMEA, and enabling of RTCM 1005 message over I2C port. Please test and let me know if it works for you.

closed 2019-04-28 by nseidle

PaulZC, 2019-06-21:
Hi Nathan ( @nseidle ),
Sincere thanks for sharing this library! I'm using it to talk to the ZED-F9P.
Can I please raise a "feature request"?
setVal() is great - but can only set one value at a time. Could you add an elegant way to set multiple keys in a single packet?
My inelegant way is to define them as a ubxPacket and send them using .sendI2cCommand : 

// setRAWXon: this is the message which enables all of the messages to be logged to SD card in one go
// It also sets the NMEA high precision mode for the GNGGA message
// It also sets the main talker ID to 'GN'
// UBX-CFG-VALSET message with key IDs of:
// 0x209102a5 (CFG-MSGOUT-UBX_RXM_RAWX_UART1)
// 0x20910232 (CFG-MSGOUT-UBX_RXM_SFRBX_UART1)
// 0x20910179 (CFG-MSGOUT-UBX_TIM_TM2_UART1)
// 0x2091002a (CFG-MSGOUT-UBX_NAV_POSLLH_UART1)
// 0x20910007 (CFG-MSGOUT-UBX_NAV_PVT_UART1)
// 0x2091001b (CFG-MSGOUT-UBX_NAV_STATUS_UART1)
// 0x10930006 (CFG-NMEA-HIGHPREC)
// 0x209100bb (CFG-MSGOUT-NMEA_ID_GGA_UART1)
// and values (rates) of 1
// 0x20930031 (CFG-NMEA-MAINTALKERID) has value 3 (GN)
static uint8_t setRAWXon_payload[] = {
  0x00, 0x01, 0x00, 0x00,
  0xa5, 0x02, 0x91, 0x20,  0x01,
  0x32, 0x02, 0x91, 0x20,  0x01,
  0x79, 0x01, 0x91, 0x20,  0x01,
  0x2a, 0x00, 0x91, 0x20,  0x00,   // Change the last byte from 0x01 to 0x00 to leave NAV_POSLLH disabled
  0x07, 0x00, 0x91, 0x20,  0x01,   // Change the last byte from 0x01 to 0x00 to leave NAV_PVT disabled
  0x1b, 0x00, 0x91, 0x20,  0x01,   // This line enables the NAV_STATUS message
  0x31, 0x00, 0x93, 0x20,  0x03,   // This line sets the main talker ID to GN
  0x06, 0x00, 0x93, 0x10,  0x01,   // This sets the NMEA high precision mode
  0xbb, 0x00, 0x91, 0x20,  0x01 }; // This (re)enables the GGA mesage
ubxPacket setRAWXon = { 0x06, 0x8a,  49, 0, 0,  setRAWXon_payload,  0, 0, false };

Sincere thanks,
Paul

nseidle, 2019-06-21:
@PaulZC I started a separate issue for this.