sparkfun/SparkFun_Ublox_Arduino_Library #11: RELPOSNED

aortner, 2019-04-24:
Could you add ubx-relposned to libaray.

It´s very interesting to build a dual antenna system to provide accurate heading and roll.

best regards andreas

nseidle, 2019-04-25:
Which fields do you want exposed and how?

[image]

I think I've got most exposed. Let me know if you need more. Please checkout **Example 5 Relative Positioning Information** for the ZED-F9P in the latest lib version.

closed 2019-04-25 by nseidle

aortner, 2019-04-25:
Wow Thank you!

I would also Need Bytes 20 and 24

Baselengh and Heading!

Best regards Andreas

nseidle, 2019-04-28:
Ok, I've added (long) Length, (long) Heading, (signed byte) HPN, (signed byte) HPE, (signed byte) HPD, (signed byte) HPLength. Please see (new) Example 5.

Please kick tires and let me know if you have issues. I'll push a new version in a few minutes so it will appear in Arduino IDE.