sparkfun/SparkFun_Ublox_Arduino_Library #12: ZED-F9P Example 4_BaseWithLCD does not output anything to Serial Port

bslosman, 2019-05-02:
I have set up the exact hardware for the Sparkfun ZED-F9P Base station receiver and loaded the sketch. The output initially included surveying in the base station and suceeded. Subsequent restarts show Base Station survey complete. After I receive the RTCM no broadcasting in the serial monitor and the LCD show "Transmitting RTCM" I am not receiving any output to the serial monitor or any changes to the LCD. I was expecting to see "pretty-print" Hex values in the serial monitor to confirm that the RTCM messages are being generated and sent. I have reset the Ublox with Example 12 and reloaded the above sketch with no change in output. I am trying to troubleshoot this before I begin sending the data to a NTRIP caster.

nseidle, 2019-05-09:
After discussing with Mark in tech support, you're right. And I think @wittend [found my error](https://forum.sparkfun.com/viewtopic.php?f=116&t=49874&p=204161&hilit=zed+fn9#p204161). We believe the problem is that in the ZED-F9P example we disable all protocols on the I2C port except for I2C. This means, even though the RTCM is being generated at output over serial, nothing is coming out of the I2C port.

As a possible solution add to the end of setup()

    myGPS.setI2COutput(COM_TYPE_UBX | COM_TYPE_RTCM3);

I'll try to confirm this shortly and update the library.

nseidle, 2019-05-14:
I found a handful of issues and am getting RTCM output now:

[image]

Should be fixed in the latest version of lib. Please install >v1.4.0. Please open new issue if you still have a problem.

closed 2019-05-14 by nseidle