sparkfun/SparkFun_Ublox_Arduino_Library #22: Definitions for VAL_LAYER need to be shifted

PaulZC, 2019-06-22:
Hi,
The bit definitions for the setVal layer field need to be shifted.
The current default of BBR actually sets the value in RAM.
The definitions in .h should be:

const uint8_t VAL_LAYER_RAM = (1 << 0);
const uint8_t VAL_LAYER_BBR = (1 << 1);
const uint8_t VAL_LAYER_FLASH = (1 << 2);

Let me know if you need a pull request.
Thanks!
Paul

nseidle, 2019-06-25:
Hi Paul - thanks for poking at this!

PRs are always welcome so I can see exactly what you're talking about.

I don't quite follow your bit shifts. This is where (I think) I got the original locations from. Page 40 of ZED-F9P interface manual.

[image]

PaulZC, 2019-06-26:
Hi Nathan,

Ah, yes, Those are the Bitfield definitions for the (deprecated) UBX-CFG-CFG message. The UBX-CFG-VALSET message has its own Bitfield definitions for the "layers" byte. See section 5.9.25 (page 88 or thereabouts).

[image]

I _think_, in the library, you only use VAL_LAYER_RAM, _BBR and _FLASH in conjunction with the CFG-VALSET message, not the CFG-CFG message? Assuming that is true, then the definitions in .h should be:

const uint8_t VAL_LAYER_RAM = (1 << 0);
const uint8_t VAL_LAYER_BBR = (1 << 1);
const uint8_t VAL_LAYER_FLASH = (1 << 2);

I'll submit a PR shortly. Once I figure out how. It'll be my first one! Yup, really. Not done this before...
Thanks,
Paul

closed 2019-06-27 by nseidle