sparkfun/SparkFun_Ublox_Arduino_Library #23: Just an observation: sendCommand will always timeout with custom ubxPackets

PaulZC, 2019-06-22:
Hi,
This is just an observation that will (hopefully) be trumped by multi-valset #20 

If you define a custom ubxPacket which holds (e.g.) a UBX-CFG-VALSET message and then send it with sendCommand, it will always timeout as the commandAck checking in processUBXpacket (called from waitForResponse \ checkUblox \ checkUbloxI2C \ process \ processUBX) expects the packet class and ID to be defined in packetCfg, not the custom packet.

The offending line is:
if (msg->id == UBX_ACK_ACK && msg->payload[0] == packetCfg.cls && msg->payload[1] == packetCfg.id)

It would be possible to correct this by passing the requestedClass and requestedID from waitForResponse further down the food chain.

Cheers,
Paul

nseidle, 2019-06-25:
I see what you mean. If you've got a proposed solution (here or on #20) I'm all ears.

PaulZC, 2019-07-01:
Strictly this Issue should probably stay open as I haven't actually fixed it with PR #28 but the problem has gone away as the new newCfgValset, addCfgValset and sendCfgValset functions for Issue #20 work on packetCfg and don't require a custom packet.

I propose we close this though as life really is too short!

Thanks!
Paul

closed 2019-12-27 by nseidle

PaulZC, 2020-04-22:
Finally resolved once and for all in #97!