sparkfun/SparkFun_Ublox_Arduino_Library #27: Most of maxWait arguments are ignored

remrol, 2019-06-28:
There are a lot of public getXXX(uint16_t maxWait = 250) methods with default maxWait set to 250. However specified maxWait argument is ignored in most cases since implementation relies on default maxWait

int32_t SFE_UBLOX_GPS::getLatitude(uint16_t maxWait)
{
  if (moduleQueried.latitude == false)
     getPVT();  <-- Here getPVT uses its default maxWait

IMHO caused by too many default arguments in helper methods like:
sendCommand, sendI2cCommand, waitForResponse, getPVT

nseidle, 2019-06-28:
That's fair. Please don't take my tone in the wrong way, but do you have a recommendation?

Do you have an issue? Can you provide the code that demonstrates this issue?

nseidle, 2019-12-23:
Easy fix.

closed 2019-12-23 by nseidle