sparkfun/SparkFun_Ublox_Arduino_Library #29: What is the resolution of getHighResLongitude?

trycoon, 2019-07-02:
What is the resolution of "int32_t longitude = myGPS.getHighResLongitude()"?
A change to the least significant digit of the longitude/latitude, how many millimeters does that change represent? So that I could measure the real life distance traveled.

Is there any formulas och examples to illustrate this?

IanCampbell-Kelly, 2019-07-03:
This looks like a question of simple mathematics?
Normal precision is 1(e-7) of a degree. My experience of this is that a u-blox 6 chip only reads the same position twice, one after the other, in about a thousand readings when the aerial is stationary. The last two digits constantly vary, even with eight or nine satellites in view.
The ublox 9 manual (which is freely available on the Web) for the "high precision geodetic position solution", UBX-NAV-HPPOSLLH says that the latitude and longitude are available to 10(e-9) of a degree, that's a billionth of a degree. Assuming that the earth is a sphere, a quarter of the circumference (90 degrees) is about 10 million metres, so a 90-billionth of this is about 0.1 millimetres. The Wikipedia articles on the GPS suggest that the system is actually much less accurate than this, a few metres.
There's no simple answer, and Google finds many discussions of GPS accuracy.

trycoon, 2019-07-03:
Yes, and some knowledge of how this library and U-blox represent longitude
and latitude in the high-resolution mode. :-)

I think the answer to my question was "10(e-9)", the less precise functions
has some kind of documentation stating the precision,
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/45e987cc5dbc63d64de07e913f3bed46e622c289/src/SparkFun_Ublox_Arduino_Library.h#L240,
but the new functions is lacking that:
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/45e987cc5dbc63d64de07e913f3bed46e622c289/src/SparkFun_Ublox_Arduino_Library.h#L258.
I wish there was a similar comment stating the precision on those lines,
and in the long run some Doxygen documentation for the most used functions
so that we don't have to dive into the code each time. ;-)

For others, see page #144 in
https://cdn.sparkfun.com/assets/learn_tutorials/8/5/6/ZED-F9P_UBX_NMEA_and_RTCM_protocols.pdf

Yes, and for better precision I would suggest you to calculate the
circumference at your latitude: https://rechneronline.de/earth-radius/

I'm aiming for a couple of centimeters for my application, meters are to
coarse for me. With a base-station and enough satellites (not just GPS) you
could get quite good results: https://www.youtube.com/watch?v=b36WFCyobpE

Thanks for your help!

// Henrik

nseidle, 2019-07-04:
Thanks @IanCampbell-Kelly. Let us know how it goes @trycoon.

closed 2019-07-04 by nseidle