sparkfun/SparkFun_Ublox_Arduino_Library #32: Put module to Sleep

andriyadi, 2019-07-13:
Is not really an issue, just asking the possibility for putting the module to sleep. I don't see any methods for doing so in the library, and as we know that we can send UBX command to make the module sleep. Is it possible to add the functionality?

Thanks.

nseidle, 2019-07-14:
A fine idea! I had not looked into power saving UBX commands before. If you know the best low power config please share it or create a PR (much appreciated).

In my brief search, the only power config I see is UBX-CFG-PWR and that only seems to put the module in a GNSS stopped state. Doesn't seem very low power but perhaps you've found something else?

andriyadi, 2019-07-15:
> A fine idea! I had not looked into power saving UBX commands before. If you know the best low power config please share it or create a PR (much appreciated).
>
> In my brief search, the only power [...]

Wow, I got a reply from the one and only Nathan Seidle @nseidle :) You do go back to your root in engineering.

So, before I create a PR, I tried to understand the source code and wrote this code. Please confirm that it suits the library:

Add a new method:
```
boolean SFE_UBLOX_GPS::setPSM(powerSavingMode psm, uint16_t maxWait) {
  packetCfg.cls = UBX_CLASS_CFG; //0x06
  packetCfg.id = UBX_CFG_PMS; //0x86
  packetCfg.len = 8;
  packetCfg.startingSpot = 0;

  //Clear packet payload
  for (uint8_t x = 0; x < packetCfg.len; x++) {
    packetCfg.payload[x] = 0;
  }

  currentPSM = psm;
  packetCfg.payload[1] = psm;

  return (sendCommand(packetCfg, maxWait));
}
```

Where `powerSavingMode` is like this:
```
enum powerSavingMode
{
  PSM_FULLPOWER,
  PSM_BALANCED,
  PSM_INTERVAL,
  PSM_1HZ,
  PSM_2HZ,
  PSM_4HZ,
  PSM_ON_OFF, //special treatment
  PSM_UNKNOWN = 0xFF
} currentPSM = PSM_BALANCED;
```

I've tried it and it works. But haven't tested yet the power consumption. Please advice.

nseidle, 2019-07-16:
Thanks! You are very kind. I've been very much enjoying doing code and design again :)

Interesting, I can't find -CFG-PMS is any of the Ublox docs but it is in a few [startup guides](https://www.u-blox.com/sites/default/files/EVK-M8BZOE_UserGuide_%28UBX-17053592%29.pdf). Do you have a interface or datasheet PDF that describes the command fully?

Your code looks great. Please include a getPSM function. Please add any new functions to keywords for highlighting.

Gearme, 2019-07-30:
Power saving options are exclusive to certain receivers. The u-blox 7 and M8 modules in particular have power management options which are described in [this official application note](https://www.u-blox.com/sites/default/files/products/documents/PowerManagement_AppNote_%28UBX-13005162%29.pdf).
The ON_OFF option is the one that is especially interesting as an addition. According to the manufacturer, it can result in significantly less current draw if the module is polled less than every 10 seconds.

Gearme, 2019-07-30:
There is at least [one library by Leonardo Bispo](https://github.com/ldab/u-blox_GNSS) which already [implements ON_OFF operation](https://github.com/ldab/u-blox_GNSS/blob/master/examples/LowPower/LowPower.ino) for these modules. However, one of the issues with that library insofar as I can see is that - ironically - it will not work with microcontroller deep sleep settings like those of the ubiquitous ESP32. An implementation of ON_OFF operation in this library should ideally be compatible with the microcontroller's own sleep functions for the least possible current draw.

nseidle, 2019-08-07:
@andriyadi - Any news of a PR? I like this feature.

tedder, 2019-09-05:
@gearme why won't the ldab library work with deep sleep?

nseidle, 2019-12-23:
This completed on PR 56. Please open a new issue if there's a feature that didn't get covered.

closed 2019-12-23 by nseidle