sparkfun/SparkFun_Ublox_Arduino_Library #37: 127 bytes available bug

nseidle, 2019-09-03:
Not sure if it's a bug, but wanted to document a weird ZED-F9P I2C comm issue:

Platform: Artemis

When requesting bytes available, we often see 0x007F or 127 bytes:

[image]

Unfortunately all these bytes are empty:

[image]

This happens 1.15s apart:

[image]

mazgch, 2019-09-16:
Could this issue be related to a mechanism of the u-blox GPS referred to as clock stretching. An I2C slave is allowed to hold down the clock if it needs to reduce the bus speed. The master on the other hand is required to read back the clock signal after releasing it to high state and wait until the line has actually gone high. 
Does Artemis support slaves that make use of clock stretching?

nseidle, 2019-10-07:
Artemis does support clock stretching. I believe you can see the ublox module stretching the lock in the image in the OP:

[image]

I believe it's something else internal to the ZED but I can't pin it down.

Further thoughts:
* I've seen similar behavior with the SAM-M8Q.
* I see fewer issues like this on the ATmega328 based systems but not zero

mazgch, 2019-10-10:
There was a similar issue in the nodeRED firmware, maybe this workaround can be applied. 
https://github.com/nodemcu/nodemcu-firmware/issues/1586

nseidle, 2020-01-08:
I haven't seen this bug in awhile. I think it's mostly been taken care of with the recent lib changes.

closed 2020-01-08 by nseidle