sparkfun/SparkFun_Ublox_Arduino_Library #38: Module not having bytes available

nseidle, 2019-09-03:
Platform: Artemis
Measurement period is extreme: 10Hz with nav rate of 5Hz (update rate of 2Hz)

Module will say bytes are available but will present 0xFFs towards the end of the frame.

In this example, 0x64 bytes are 'available':

[image]

But part of the way through the byte harvesting, we start getting 0xFFs:

[image]

And then our CRC is corrupt. This error occurs ~1 per 60 seconds.

Often the frame will include 0xFFs in between what looks to be valid data:

[image]

It seems as though the module is busy updating other ports (serial, SPI, etc) and is not serving up bytes to I2C buffer.

nseidle, 2019-09-04:
I've updated my ZED-F9P to UBX_F9_100_HPG_112_ZED_F9P firmware and have the same issue.

I2C rate is 400kHz.

mazgch, 2019-09-16:
Could this issue be related to a mechanism of the u-blox GPS referred to as clock stretching. An I2C slave is allowed to hold down the clock if it needs to reduce the bus speed. The master on the other hand is required to read back the clock signal after releasing it to high state and wait until the line has actually gone high. 
Does Artemis support slaves that make use of clock stretching?

mazgch, 2019-10-31:
Have you disabled the UART1 out protocol before switching to high nav rate. I like to avoid that Issue in the I2C is caused by the overload on other ports ports such as the UART. The receiver may run low on memory because the buffer queue up due to low baud rate on the UART (even it is not used). 

So can you try to disable UART1/2/USB out protocol with CFG-PORT before switching to high rates. 

I will try again to reproduce Example6_OutputRate here.

mazgch, 2019-11-01:
Count you please disable the output before switching to 10 Hz so around line 49 in the Example6_OutputRate. 
```
myGPS.setUART1Output(0); //Set the UART1 port output 
myGPS.setUART2Output(0); //Set the UART2 port output
myGPS.setI2COutput(COM_TYPE_UBX); //Set the I2C port to output UBX only (turn off NMEA noise)
myGPS.setNavigationFrequency(10); //Set output to 10 times a second
```

If I do this I see stable 10Hz in the serial monitor. I have not checked with the logic analyzer.

sslupsky, 2020-05-22:
@nseidle I understand the FF's are inserted by the module to indicate there is no data available to send.  I noticed you had a comment about that here
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/61c85b6509040159c57463276b182f3782043f72/src/SparkFun_Ublox_Arduino_Library.cpp#L311

It is not a bug with the module but expected behaviour.

nseidle, 2020-05-22:
Yes and no. The preamble includes the number of bytes that should be available. These are read out and, at times, include 0xFFs. This is the bug.

sslupsky, 2020-05-22:
I think the module might still be indicating that data is not available?  That is, you are reading too fast?  Try re-reading the last byte and see if the checksum works out.  If that is the case, then it is likely that it just can't keep up.

sslupsky, 2020-05-22:
The module likely runs time sensitive functions from time to time.  If you are reading when one of these things is going on, it might just defer the read.

sslupsky, 2020-05-22:
On second thought, that would violate their specification.  They only place they are supposed to insert the FF is at the beginning of the message to indicate it is not a valid message/not ready.

Did you file a bug report with ublox about this?  I think you are correct, this looks like a bug.

PaulZC, 2020-05-24:
I'm wondering if this issue is related to pull-ups too?
Please see my comment on #40.
Food for thought?
Cheers,
Paul

nseidle, 2020-05-26:
I think we've solved most of this issue by removing the pull up resistors. If you're experiencing anything like this be sure to remove all pullups on peripheral boards and or the controller. Most, if not all, u-blox modules have built in pull ups.

Notes added to main readme.

closed 2020-05-26 by nseidle