sparkfun/SparkFun_Ublox_Arduino_Library #41: uBlox MAX-7Q returning zeros

Patrik657, 2019-10-09:
### Subject of the issue
Hi,
I have connected uBlox MAX-7Q to Arduino Nano. Using this library and example "Example11_UseUart" from this library I was able to connect uBlox to Arduino, but all I receive in return is zeros.
Do you have any ideas what I did wrong, did I forget to define/uncomment in .h file, is something wrong with my GPS module?...
Thanks for replies in advance.

### Your workbench
Arduino Nano v3.3
* What version of hardware or breakout board are you using? 
* How is the breakout board wired to your microcontroller?
* How is everything being powered?
* Are there any additional details that may help us help you?

### Steps to reproduce

// Hello LoRa - ABP TTN Packet Sender (Multi-Channel)
// Tutorial Link: https://learn.adafruit.com/the-things-network-for-feather/using-a-feather-32u4
//
// Adafruit invests time and resources providing this open source code.
// Please support Adafruit and open source hardware by purchasing
// products from Adafruit!
//
// Copyright 2015, 2016 Ideetron B.V.
//
// Modified by Brent Rubell for Adafruit Industries, 2018

/************************** Configuration ***********************************/

#include <TinyLoRa.h>
#include <SPI.h>

/*
  Reading lat and long via UBX binary commands using UART @38400 baud - free from I2C
  By: Nathan Seidle, Adapted from Example3_GetPosition by Thorsten von Eicken
  SparkFun Electronics
  Date: January 28rd, 2019
  License: MIT. See license file for more information but you can
  basically do whatever you want with this code.
  This example shows how to configure the library and U-Blox for serial port use as well as
  switching the module from the default 9600 baud to 38400.
  Note: Long/lat are large numbers because they are * 10^7. To convert lat/long
  to something google maps understands simply divide the numbers by 10,000,000. We 
  do this so that we don't have to use floating point numbers.
  Leave NMEA parsing behind. Now you can simply ask the module for the datums you want!
  Feel like supporting open source hardware?
  Buy a board from SparkFun!
  ZED-F9P RTK2: https://www.sparkfun.com/products/15136
  NEO-M8P RTK: https://www.sparkfun.com/products/15005
  SAM-M8Q: https://www.sparkfun.com/products/15106
  Hardware Connections:
  Connect the U-Blox serial TX pin to Uno pin 10
  Connect the U-Blox serial RX pin to Uno pin 11
  Open the serial monitor at 115200 baud to see the output
*/

#include "SparkFun_Ublox_Arduino_Library.h" //http://librarymanager/All#SparkFun_Ublox_GPS
SFE_UBLOX_GPS myGPS;

#include <SoftwareSerial.h>
SoftwareSerial mySerial(0, 1); // RX, TX. Pin 10 on Uno goes to TX pin on GPS module.
SoftwareSerial Serial_PC(4, 5); // RX, TX

long lastTime = 0; //Simple local timer. Limits amount of I2C traffic to Ublox module.

void setup()
{
  Serial_PC.begin(115200);
  //while (!Serial_PC); //Wait for user to open terminal
  Serial_PC.println("SparkFun Ublox Example");

  //Assume that the U-Blox GPS is running at 9600 baud (the default) or at 38400 baud.
  //Loop until we're in sync and then ensure it's at 38400 baud.
  do {
    Serial_PC.println("GPS: trying 38400 baud");
    mySerial.begin(38400);
    if (myGPS.begin(mySerial) == true) break;

    delay(100);
    Serial_PC.println("GPS: trying 9600 baud");
    mySerial.begin(9600);
    if (myGPS.begin(mySerial) == true) {
        Serial_PC.println("GPS: connected at 9600 baud, switching to 38400");
        myGPS.setSerialRate(38400);
        delay(100);
    } else {
        //myGPS.factoryReset();
        delay(2000); //Wait a bit before trying again to limit the Serial output
    }
  } while(1);
  Serial_PC.println("GPS serial connected");

  myGPS.setUART1Output(COM_TYPE_UBX); //Set the UART port to output UBX only
  myGPS.setI2COutput(COM_TYPE_UBX); //Set the I2C port to output UBX only (turn off NMEA noise)
  myGPS.saveConfiguration(); //Save the current settings to flash and BBR
}

void loop()
{
  //Query module only every second. Doing it more often will just cause I2C traffic.
  //The module only responds when a new position is available
  if (millis() - lastTime > 3000)
  {
    lastTime = millis(); //Update the timer
    
    long latitude = myGPS.getLatitude();
    Serial_PC.print(F("Lat: "));
    Serial_PC.print(latitude);

    long longitude = myGPS.getLongitude();
    Serial_PC.print(F(" Long: "));
    Serial_PC.print(longitude);
    Serial_PC.print(F(" (degrees * 10^-7)"));

    long altitude = myGPS.getAltitude();
    Serial_PC.print(F(" Alt: "));
    Serial_PC.print(altitude);
    Serial_PC.print(F(" (mm)"));

    byte SIV = myGPS.getSIV();
    Serial_PC.print(F(" SIV: "));
    Serial_PC.print(SIV);

    Serial_PC.println();
  }
}

### Expected behavior
I was expecting to get some values, but none...

### Actual behavior
[image]

tedder, 2019-11-19:
It'd be easier to read if you formatted the code, but I assume that's an unchanged use of the example code. This is how things will look if if you don't have a "lock", common when the ublox is inside a building. You can take it outside to get a lock and/or enable debugging (`myGPS.enableDebugging(Serial_PC)`) to see what's actually coming in.

Patrik657, 2019-11-20:
Thanks for answering.
Yea I know, the code looks bad, but I am new to GitHub so I did not knew, how the code will look like or what should I do to make it look like it should. So I get one pass for that :)

Now about the problem: I tried and used another library for uBlox (TinyGPSplus by SensorsIoT), you can find it on this link: https://github.com/SensorsIot/TinyGPSplus-for-Neo-M8N
With TinyGPS library I was able to get all the info from the module, so I proceeded my project with that library.
So I still do not know, what was wrong with this library.

Thanks for answering anyway.

nseidle, 2019-11-20:
Thanks tedder, I agree. Patrik - no worries, gists are great for code in the future.

Also, I attempted to get a hold of a MAX-7Q to try to replicate this issue but this is an older Ublox product and is sparsely available. I don't think we can help much since we can't replicate the hardware.

Good to hear TinyGPS is working for you. I suspect it has something to do with our UART interface. This lib works with serial but I've done the vast majority of my work using the I2C interface. I have tested the serial interface much less. There might be something wonky with serial + MAX-7Q + this lib but I can't really prove anything without hardware.

I'm going to close this issue. If anyone has thoughts or similar issues, please re-open.

closed 2019-11-20 by nseidle