sparkfun/SparkFun_Ublox_Arduino_Library #44: millis() does not work during sleep

sslupsky, 2019-10-19:
### Subject of the issue
checkUbloxI2C() uses millis() to check time and this time can be unreliable if the mcu goes to sleep since the SysTick on which millis() is based is stopped during sleep.  Since checkUbloxI2C() uses a global "lastcheck" to track time elapsed between calls to checkUbloxI2C(), this causes an undesirable additional delay when checking the module after waking from sleep resulting in increased power consumption.

millis() is also used in waitForResponse() but since that function only uses millis() for relative time elapsed within the function itself, it is not a problem here except that generally, it would be preferable if waitForResponse() was not blocking.

My workaround for this problem was to use a tick from the real time clock instead of millis().  In the code snippet below, I have replaced the call to millis() with a global variable (volatile) that keeps track of the time from the rtc.

I suggest perhaps the "proper" solution might be abstract millis() using a virtual function and if the user needs to replace it with something else, it can be replaced by instantiating your own millis() function.

```
boolean SFE_UBLOX_GPS::checkUbloxI2C()
{
  if (rtcTickMS - lastCheck >= I2C_POLLING_WAIT_MS)
  {
    //Get the number of bytes available from the module
    uint16_t bytesAvailable = 0;
    _i2cPort->beginTransmission(_gpsI2Caddress);
    _i2cPort->write(0xFD);                     //0xFD (MSB) and 0xFE (LSB) are the registers that contain number of bytes available
    if (_i2cPort->endTransmission(false) != 0) //Send a restart command. Do not release bus.
      return (false);                          //Sensor did not ACK

    _i2cPort->requestFrom((uint8_t)_gpsI2Caddress, (uint8_t)2);
    if (_i2cPort->available())
    {
      uint8_t msb = _i2cPort->read();
      uint8_t lsb = _i2cPort->read();
      bytesAvailable = (uint16_t)msb << 8 | lsb;
    }

    if (bytesAvailable == 0)
    {
      if (_printDebug == true)
      {
        _debugSerial->println("No bytes available");
      }
      lastCheck = rtcTickMS; //Put off checking to avoid I2C bus traffic
      return true;
    }

```

I also noticed today commit e3e170ea made in July to address an issue when the module is not ready.  The commit adds a blocking delay in this event which causes additional power consumption.  I would like to ask if there is something that can be done to eliminate the blocking delay.

nseidle, 2019-11-20:
I agree millis() is not ideal for low power applications but it's what is guaranteed to be available. I like the idea of an RTC source but very few projects are going to have that intrinsically. 

I'm very open to ideas and solutions. Polling versus blocking is fine by me. isAvailable() or available() is the classic that I enjoy.

sslupsky, 2020-01-27:
I think perhaps that time should be a parameter of the checkUbloxI2C() function?  That is, the caller provides the current time and then the function uses that to determine the elapsed time since it checked last time.

waitForAckResponse(), waitForNoAckResponse() and getPVT() would also require the current time to be passed into them since these functions call checkUblox().

nseidle, 2020-01-27:
It's an interesting idea. If you'd like to propose a separate PR (from your getPowerxxx work as well) with an example, I'd consider it.

PaulZC, 2020-04-24:
It looks as if this issue has gone stale?
I am going to close it but please reopen it again if this needs further discussion.
Cheers,
Paul

closed 2020-04-24 by PaulZC