sparkfun/SparkFun_Ublox_Arduino_Library #58: Add getPowerState method

nseidle, 2019-12-23:
We just added powerSaveMode(). Would be nice to have a getPowerState method to see what state the module is in.

sslupsky, 2020-01-27:
I did some work on this.  I added some functions:

getPowerMode()
setPowerMode()
sleep()
getExtPMConfig()
setExtPMConfig()

Let me know if you are interested in this approach to the API and I can submit a PR.

nseidle, 2020-01-27:
Absolutely! Thanks. Please checkout our [contributing](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/master/CONTRIBUTING.md) doc for style guides.

sslupsky, 2020-01-27:
Submitted a PR.  Let me know what you think.  I do not have an example at the moment.  I used these a few times when I was working on making the module go into its low power state.

I included a couple other functions as well:

setPortTXReady()
This is useful for saving power.

pollPVT()
Allows you to do a non blocking getPVT() and then poll.

engelberger5353, 2020-04-17:
Hi, I have desperately tried to send the RXM-PMREQ command to my zed f9p, however I receive a timeout, I am interested in your changes to see if it is possible to understand why it does not work.

PaulZC, 2020-04-22:
Added getPowerSaveMode in #99.

Closing...

closed 2020-04-22 by PaulZC