sparkfun/SparkFun_Ublox_Arduino_Library #67: I2C/RTCM issues

eviltwinkie, 2020-01-08:
### Subject of the issue
ESP32 - I2C Messages timeout and rtcmFrameCounter keeps incrementing until panic occurs.
SAMD21 - I2C will not work unless I change #define I2C_BUFFER_LENGTH 255 as 256 will not work.

### Your workbench
ESP32 (Sparkfun ESP32 1 Channel Gateway)/SAMD21 (Adafruit M0) both using I2C and similar sketches. Adafruit is more stable.

### Steps to reproduce
Update all libs to latest version.
Load "Example3_StartRTCMBase" with debugging enabled on either platform.
ESP32 partially works. M0 will not work without changing the ring buffer size.

### Expected behavior
Packets should be processed appropriately, ACKs received and the frame counter reset. 

### Actual behavior
Timeouts should not take 35 seconds. Eventually memory leak will stop the RTCM output or panic occurs.

See below debug.

Sending: CLS:CFG ID:8A Len: 0xA Payload: 0 1 0 0 BD 2 91 20 4 0
sendCommand: Waiting for ACK response
checkUbloxI2C: Large packet of 24040 bytes received
Incoming: Size: 9 Received: CLS:CFG ID:8B Len: 0x9 Payload: 1 1 0 0 BD 2 91 20 4
packetCfg now valid
Incoming: Size: 2 Received: CLS:ACK ID:1 Len: 0x2 Payload: 6 8B
packetCfg now valid
packetAck now valid
UBX ACK: Command sent/ack'd successfully
Module not ready with data
Module not ready with data
Incoming: Size: 2 Received: CLS:ACK ID:1 Len: 0x2 Payload: 6 8A
packetCfg now valid
packetAck now valid
waitForACKResponse: ACK received after 40664 msec
waitForACKResponse: New data successfully sent

Sending: CLS:CFG ID:SAVE Len: 0xC Payload: 0 0 0 0 FF 0 0 0 0 0 0 0
sendCommand: Waiting for ACK response
checkUbloxI2C: Large packet of 21557 bytes received
Module not ready with data
Incoming: Size: 2 Received: CLS:ACK ID:1 Len: 0x2 Payload: 6 9
packetAck now valid
UBX ACK: Command sent/ack'd successfully
waitForACKResponse: ACK received after 36503 msec
waitForACKResponse: New data successfully sent
 OK
CONFIG: CFG-MSGOUT-RTCM_3X_TYPE1077_I2C (0x209102cc)key: 0x209102CC

Sending: CLS:CFG ID:8B Len: 0x8 Payload: 0 1 0 0 CC 2 91 20
sendCommand: Waiting for ACK response
checkUbloxI2C: Large packet of 21197 bytes received
waitForACKResponse: timeout after 35795 msec. No ack packet received.

Sending: CLS:CFG ID:8A Len: 0xA Payload: 0 1 0 0 CC 2 91 20 4 0
sendCommand: Waiting for ACK response
checkUbloxI2C: Large packet of 22026 bytes received
Incoming: Size: 9 Received: CLS:CFG ID:8B Len: 0x9 Payload: 1 1 0 0 CC 2 91 20 4
packetCfg now valid
Incoming: Size: 2 Received: CLS:ACK ID:1 Len: 0x2 Payload: 6 8A
packetCfg now valid
packetAck now valid
waitForACKResponse: Config was valid but ACK not received
waitForACKResponse: timeout after 37168 msec. No ack packet received.

Sending: CLS:CFG ID:SAVE Len: 0xC Payload: 0 0 0 0 FF 0 0 0 0 0 0 0
sendCommand: Waiting for ACK response
checkUbloxI2C: Large packet of 22078 bytes received
Module not ready with data
Incoming: Size: 2 Received: CLS:ACK ID:1 Len: 0x2 Payload: 6 9
packetAck now valid
UBX ACK: Command sent/ack'd successfully
waitForACKResponse: ACK received after 36866 msec
waitForACKResponse: New data successfully sent
 OK
CONFIG: CFG-MSGOUT-RTCM_3X_TYPE1087_I2C (0x209102d1)key: 0x209102D1

Sending: CLS:CFG ID:8B Len: 0x8 Payload: 0 1 0 0 D1 2 91 20
sendCommand: Waiting for ACK response
checkUbloxI2C: Large packet of 22820 bytes received
waitForACKResponse: timeout after 38205 msec. No ack packet received.

Sending: CLS:CFG ID:8A Len: 0xA Payload: 0 1 0 0 D1 2 91 20 4 0
sendCommand: Waiting for ACK response
checkUbloxI2C: Large packet of 22224 bytes received
Incoming: Size: 9 Received: CLS:CFG ID:8B Len: 0x9 Payload: 1 1 0 0 D1 2 91 20 4
packetCfg now valid
Module not ready with data
Incoming: Size: 2 Received: CLS:ACK ID:1 Len: 0x2 Payload: 6 8A
packetCfg now valid
packetAck now valid
waitForACKResponse: Config was valid but ACK not received
waitForACKResponse: timeout after 37917 msec. No ack packet received.

Sending: CLS:CFG ID:SAVE Len: 0xC Payload: 0 0 0 0 FF 0 0 0 0 0 0 0
sendCommand: Waiting for ACK response
checkUbloxI2C: Large packet of 21402 bytes received
Module not ready with data
Incoming: Size: 2 Received: CLS:ACK ID:1 Len: 0x2 Payload: 6 9
packetAck now valid
UBX ACK: Command sent/ack'd successfully
waitForACKResponse: ACK received after 36370 msec
waitForACKResponse: New data successfully sent
 OK
CONFIG: CFG-MSGOUT-RTCM_3X_TYPE1097_I2C (0x20910318)key: 0x20910318

Sending: CLS:CFG ID:8B Len: 0x8 Payload: 0 1 0 0 18 3 91 20
sendCommand: Waiting for ACK response
checkUbloxI2C: Large packet of 21014 bytes received
waitForACKResponse: timeout after 35516 msec. No ack packet received.

Sending: CLS:CFG ID:8A Len: 0xA Payload: 0 1 0 0 18 3 91 20 4 0
sendCommand: Waiting for ACK response
checkUbloxI2C: Large packet of 22325 bytes received
Incoming: Size: 9 Received: CLS:CFG ID:8B Len: 0x9 Payload: 1 1 0 0 18 3 91 20 4
packetCfg now valid
Incoming: Size: 2 Received: CLS:ACK ID:1 Len: 0x2 Payload: 6 8B
packetCfg now valid
packetAck now valid
UBX ACK: Command sent/ack'd successfully
Incoming: Size: 2 Received: CLS:ACK ID:1 Len: 0x2 Payload: 6 8A
packetCfg now valid
packetAck now valid
waitForACKResponse: ACK received after 37699 msec
waitForACKResponse: New data successfully sent

Sending: CLS:CFG ID:SAVE Len: 0xC Payload: 0 0 0 0 FF 0 0 0 0 0 0 0
sendCommand: Waiting for ACK response
checkUbloxI2C: Large packet of 22506 bytes received
waitForACKResponse: timeout after 38125 msec. No ack packet received.
 OK
CONFIG: CFG-MSGOUT-RTCM_3X_TYPE1127_I2C (0x209102d6)key: 0x209102D6

Sending: CLS:CFG ID:8B Len: 0x8 Payload: 0 1 0 0 D6 2 91 20
sendCommand: Waiting for ACK response
checkUbloxI2C: Large packet of 22447 bytes received
Incoming: Size: 2 Received: CLS:ACK ID:1 Len: 0x2 Payload: 6 9
packetAck now valid
waitForACKResponse: timeout after 38223 msec. No ack packet received.

Sending: CLS:CFG ID:8A Len: 0xA Payload: 0 1 0 0 D6 2 91 20 4 0
sendCommand: Waiting for ACK response
checkUbloxI2C: Large packet of 24452 bytes received
Incoming: Size: 9 Received: CLS:CFG ID:8B Len: 0x9 Payload: 1 1 0 0 D6 2 91 20 4
packetCfg now valid
Incoming: Size: 2 Received: CLS:ACK ID:1 Len: 0x2 Payload: 6 8B
packetCfg now valid
packetAck now valid
UBX ACK: Command sent/ack'd successfully
Module not ready with data
Incoming: Size: 2 Received: CLS:ACK ID:1 Len: 0x2 Payload: 6 8A
packetCfg now valid
packetAck now valid
waitForACKResponse: ACK received after 41442 msec
waitForACKResponse: New data successfully sent

nseidle, 2020-01-08:
This sounds very similar to the [fix we put in place](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/commit/34a907c01862b7c05d99cba76b94605b80fd647b#diff-09b68aa0742e5325ae8fb037d77970ebR642) a few days ago. If the packet length gets corrupt we can potential load more bytes than the payload can handle.

If the rtcmLen variable gets corrupt then we have to wait until for (rtcmFrameCounter == rtcmLen) for the completion of a RTCM frame; this could by up to uint16_t or 65k bytes. I believe RTCM frames are limited to ~1k max (just off the top of my head here) so we could put a sanity check in there... 

Can you give [this branch](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/tree/fixRTCMframeCounter) a try and let me know if it works any better? I don't know what you're doing within processRTCM but that too may be a culprit.

I *believe* Adafruit defines the I2C ring buffer differently on their trinket than other SAMD21 based boards. Not sure. But I've changed the I2C_BUFFER_LENGTH define to:

#define I2C_BUFFER_LENGTH (SERIAL_BUFFER_SIZE - 1)

In an attempt to hit your 255 issue.

eviltwinkie, 2020-01-09:
I'll try the new branch and let you know. I think I also came to the conclusion that its a processRTCM problem. I'm checking the frame counter for the "1" and then doing an action. If I take too long in processing, bad things start to happen all over the place.

So I'm going to implement a ring buffer to handle it via a second thread so it doesn't block or interrupt the processRTCM function.

eviltwinkie, 2020-01-10:
The panic's have reduced but I'm still seeing some issues. Also, still hitting the counter.

processUBX: counter hit MAX_PAYLOAD_SIZE

I've also for sure narrowed down that some of the issue was due to the processing time while inside the processRTCM function. Using a tasks and queues I could see the queue get filled pretty quick. After it filled up, Bad Things...

Definitely some sort of locking/blocking/ISR issue. I might move the entire SFE_UBLOX_GPS::process into its own task and give it a queue to keep that on its own processor.

PaulZC, 2020-04-24:
Hi @eviltwinkie ,
Can you please test V1.8.0 of the library as I think it will resolve most of your issues?
I am going to close this issue but please do reopen it again if it needs further discussion.
Thanks for contributing!
Best wishes,
Paul

closed 2020-04-24 by PaulZC