sparkfun/SparkFun_Ublox_Arduino_Library #69: Zephyr driver

sslupsky, 2020-01-27:
### Subject of the issue
This should probably be someplace else but I am not sure where to ask it.

I've been doing some work with the Zephyr RTOS and wondering if Sparkfun is considering porting some of their drivers to Zephyr?  If so, I could help out with some testing.

I think the current Arduino driver would port over reasonably well.  Though this driver is written in  C++ and Zephyr drivers need to be written in C, there isn't much dependency on C++ that I can see.  The I2C and UART API's are a bit different but I think this would not be too much problem.

The bigger thing to consider is if you choose to do this you would be the first to develop a GPS driver for Zephyr.  So, likely there would be a need to get consensus with the Zephyr guys on an GPS API.  Some of that work is already done (data types for GPS "sensors" are defined).  But a GPS sensor API does not exist to my knowledge. 

IMHO, Zephyr has some pretty interesting potential.  Still kinda rough around the edges but it is pretty slick.

nseidle, 2020-04-15:
Hi @sslupsky - We welcome support for additional platforms. Indeed, Zephyr looks pretty neat! If anyone wants to wants to port our Ublox lib to pure C we are happy to consider a shift in that direction.

closed 2020-04-15 by nseidle

vid553, 2020-11-02:
Hi,
I saw this issue for a Zephyr version of the library, so also writing a comment here.

I have created a pull request #145 that ports basic functionalities of this library to Zephyr OS.
Note that it doesn't have a GPS API and I left it in C++, so it is not a proper solution as described by @sslupsky, but it provides a working port to this platform.