sparkfun/SparkFun_Ublox_Arduino_Library #73: getProtocolVersion overruns payloadCfg array

matthewgkerr, 2020-02-12:
payloadCfg currently has a defined size of 256 (MAX_PAYLOAD_SIZE) bytes. Base on the comment "We need ~220 bytes for getProtocolVersion on most ublox modules" I believe the expectation is most modules will contain no more than 6 extensions (payload: 30 + 10 + 6*30 = 220 bytes). getProtocolVersion is currently configured to step through 10 extensions (340 bytes) which will overrun payloadCfg.

mayopan, 2020-02-17:
Hi, matthewgkerr.
I found the same issue, and fixed it by a little modification.
Please try [My repo](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/compare/master...mayopan:master) for getProtocolVersion.
If it works in your circumstance, I'll make PR for it.

I think the official master version has another bug yet.
I can't change baud rate or other setting changes.OR
So I have to arrange the baud rate to that of connected public gps.
This bug lies in the new method of "ack", I think.
I'm looking in it now.

mayopan, 2020-02-17:
And one more thing.
I want to know not only the version number, but also all other extension information of ublox gps module, as such you can see in U-center software.
So I made an additional function "getModuleInfo".
It outputs the text arrays as named structure.
[Try it](https://github.com/mayopan/SparkFun_Ublox_Arduino_Library/tree/develop/examples) too, if you are interested.

fm4dd, 2020-04-09:
> I want to know not only the version number, but also all other extension information of ublox gps module, as such you can see in U-center software.
> So I made an additional function "getModuleInfo".

Looks great! You should create a PR for it. 

Example run on a SAM-M8Q:
```
Module Info : 
Soft version: ROM CORE 3.01 (107888)
Hard version: 00080000
Extensions:
  FWVER=SPG 3.01
  PROTVER=18.00
  GPS;GLO;GAL;BDS
  SBAS;IMES;QZSS

Done!
```

PaulZC, 2020-04-22:
Hi @matthewgkerr & @mayopan ,
We need to be careful that we don't increase the RAM usage for most users. The library already needs more than half of the available RAM on ATmega328P platforms. So I'm afraid increasing the default buffer size inside the library is not going to be possible.
But, if your platform has RAM to spare, there are now two possible ways forward without modifying the library:
- ```#define MAX_PAYLOAD_SIZE 512``` _before_ the ```#include "SparkFun_Ublox_Arduino_Library.h"``` in your .ino. This will override the MAX_PAYLOAD_SIZE and create extra space for the module info.
- Use the new custom packet feature to define your own packet with a larger payload in your .ino. Please see the new [Example20](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/master/examples/Example20_SendCustomCommand/Example20_SendCustomCommand.ino) for details.

I am going to close this issue but please reopen it if you are still having problems.
Enjoy!
_**Paul**_

closed 2020-04-22 by PaulZC

mayopan, 2020-05-15:
> Looks great! You should create a PR for it.
>
> Example run on a SAM-M8Q:
>
> ```
> Module Info :
> Soft version: ROM CORE 3.01 (107888)
> Hard version: 00080000
> Extensions:
>   FWVER=SPG 3.01
>   PROTVER=18.00 [...]

The new example21 has been added in [the latest commit](814da75592477e4d60260e80aa9bc7958bf2f3e7)   of original library.
It shows how to use the same function getModuleInfo without library modification.
Enjoy!

nseidle, 2020-09-26:
Hi @mayopan - just wanted to say thanks for your contribution! I needed to grab hwVersion for one of my projects and your solution and example are perfect.