sparkfun/SparkFun_Ublox_Arduino_Library #80: Add support for the Neo 6M chip (can send PR if you want)

geeksville, 2020-03-14:
### Subject of the issue
Are you interested in support for the older 6M ublox chip?  If so, I can send in a PR.  The existing code in this repo mostly works, but the 6M doesn't have the PVT nav commands.  So I needed to generalize it to use PVT on the newer chips, but fall back to TIMEUTC or POSLLH on the 6M.

My (I would cleanup a bit if you want this PR) change is here:

https://github.com/meshtastic/SparkFun_Ublox_Arduino_Library/commit/512833eef5124226a36747f19a320f6aee4b53e6

nseidle, 2020-03-27:
We welcome additions. I'm not too enthused with adding 6M support since it's older but if it's a reasonable merge, we can do it.

closed 2020-04-15 by nseidle

tha23rd, 2020-06-01:
@geeksville did you ever go about making said PR? I'm also looking for compatibility with the 6M. Thanks!

geeksville, 2020-06-01:
I did not but if you pull from here I bet it will work great for you:
https://github.com/meshtastic/SparkFun_Ublox_Arduino_Library

(Sent from a phone - please ignore typos)

tha23rd, 2020-06-01:
Thanks for sharing, any details on what's changed in that fork? Will I need to make any modifications to my code for it work with a 6M?

Thanks!

geeksville, 2020-06-02:
mainly the newer PVT style reads are not supported by the 6M.  So on that
platform I use some existing apis (which will also work on the 8M).

The changes are guarded by a flag which is currently hardwired to true:

bool neo6M = true;

tha23rd, 2020-06-02:
Gotcha, thanks for the explanation - I'm still unable to get read the lat+long using your fork

My impl:
https://gist.github.com/tha23rd/0da57a3f08ba1309efadbf5da0e8f763

I have confirmed I can parse the NMEA directly using the TinyGPS++ lib, so I'm positive it isn't a pin issue. I am using a 6M clone: (https://www.amazon.com/gp/product/B084MK8BS2/ref=ppx_yo_dt_b_asin_title_o05_s01?ie=UTF8&psc=1). So I do wonder if the clone just isn't compatible.

Thanks for the help so far

geeksville, 2020-06-02:
There are some fakes out there that just do nema and they slap a ublox
sticker on.

(Sent from a phone - please ignore typos)