sparkfun/SparkFun_Ublox_Arduino_Library #85: SFE_UBLOX_STATUS_CRC_FAIL errors with direct UART connection

jafrado, 2020-04-16:
### Subject of the issue
SFE_UBLOX_STATUS_CRC_FAIL errors with direct UART connection from NEO-M8P-2 to Feather 32u4 basic TX/RX (null-modem). 

### Your workbench

Adafruit Feather 32u4 basic proto with TX/RX connected directly to NEO-M8P-2
https://learn.adafruit.com/adafruit-feather-32u4-basic-proto/pinouts

* Use https://www.sparkfun.com/products/15005 NEO-M8P-2 RTK GPS Module
* Use https://www.adafruit.com/product/2771 Atmega32u4 @8Mhz
* Use https://www.adafruit.com/product/3463 NXP 9DOF
* RX/TX (Arduino Serial2) connected to TX/RX (null-modem) of NEO-M8P2 with 3" (8cm) wire
* Use 26AWG flexible silicon wire for attaching the digital signals and power (UART, I2C, 3V3) as below:
* Bring out 3V3/GND wires from Feather 32U4 to receive 3V power from the onboard Feather power adapter.
* Connect GPS 3V3 and GND to Feather 32U4 3V3/GND Pair 2
* Connect Feather 32U4 TX to GPS RX, 32U4 RX to GPS TX
* Power is provided by Feather onboard 500mA peak AP2112K-3.3.

### Steps to reproduce
Easy sample here: https://github.com/jafrado/ubxgps, or using the latest master Sparkfun Ublox Arduino Library, and \examples\Example12_UseUart with the HW Serial Port on Atmega32u4 (Arduino Serial1, Pins 0=RX, 1=TX to NEO Pin RX/TX (Null-Modem). This means comment out the lines
```
#include <SoftwareSerial.h>
SoftwareSerial mySerial(10, 11); // RX, TX. Pin 10 on Uno goes to TX pin on GPS module.
```
now search and replace mySerial with Serial1. 

At the bottom of setup()  call myGPS.enableDebugging(Serial);

NOTE: Feather 32u4 uses USB console as Serial and the Pins 0/1 UART as Serial1

Upload to Feather and you will get the below with no debug output:
```
SparkFun Ublox Example
GPS: trying 38400 baud
GPS serial connected
Current update rate:1
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0 Fix: No fix RTK: 0 Speed: 0 (mm/s) Heading: 0 (degrees * 10^-5) pDOP: 0.00
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0 Fix: No fix RTK: 0 Speed: 0 (mm/s) Heading: 0 (degrees * 10^-5) pDOP: 0.00
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0 Fix: No fix RTK: 0 Speed: 0 (mm/s) Heading: 0 (degrees * 10^-5) pDOP: 0.00
Lat: 0 Long: 0 (degrees * 10^-7) Alt: 0 (mm) SIV: 0 Fix: No fix RTK: 0 Speed: 0 (mm/s) Heading: 0 (degrees * 10^-5) pDOP: 0.00
```

With debug output, you see the CRC Fail message over and over.
```

SparkFun Ublox Example
GPS: trying 38400 baud
GPS serial connected
Current update rate:1
own Status

Sending: CLS:NAV ID:PVT Len: 0x0 Payload:

ut
getPVT retVal: CRC Fail
Lat: 0own Status

Sending: CLS:NAV ID:PVT Len: 0x0 Payload:

Incoming: Size: 92 Received: CLS:FF ID:FF Len: 0x5C Payload: 80 AA 70 14 E4 7 4 F 17 F A 37 2 0 0 0 4D FF FE FF 3 43 EA D A6 73 7D B7 6B D2 1C 16 9 50 1 0 73 C5 1 0 79 0 0 0 84 0 0 0 11 0 0 0 F6 FF FF FF A 0 0 0 13 0 0 0 37 65 2A 1 D 0 0 0 7D B6 25 0 A0 0 0 0 12 19 26 3B 0 0 0 0 0 0 0 0

ut
getPVT retVal: CRC Fail
 Long: 0 (degrees * 10^-7)own Status

Sending: CLS:NAV ID:PVT Len: 0x0 Payload:

 checksumA: 104 checksumB: 174 rollingChecksumA: 100 rollingChecksumB: 209
Failed  : Size: 0 Received: CLS:FF ID:FF Len: 0x0 Payload:
ut
getPVT retVal: CRC Fail
 Alt: 0 (mm)own Status

Sending: CLS:NAV ID:PVT Len: 0x0 Payload:

Incoming: Size: 92 Received: CLS:FF ID:FF Len: 0x5C Payload: 50 B2 70 14 E4 7 4 F 17 F C 37 2 0 0 0 17 FE FE FF 3 43 EA D A5 73 7D B7 6D D2 1C 16 E 50 1 0 78 C5 1 0 A6 0 0 0 8C 0 0 0 13 0 0 0 FA FF FF FF 2 0 0 0 14 0 0 0 37 65 2A 1 A 0 0 0 99 5 1A 0 A0 0 0 0 12 19 26 3B 0 0 0 0 0 0 0 0

ut
getPVT retVal: CRC Fail
 SIV: 0
own Status
```

### Sample
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue. Right here, with full output excerpts and other great snippets from examples provided with the library. Sample here: https://github.com/jafrado/ubxgps

### Expected behavior
Tell us what should happen
```
g: Size: 92 Received: CLS:FF ID:FF Len: 0x5C Payload: 30 52 73 14 E4 7 4 F 17 12 4 37 2 0 0 0 5C 96 FE FF 3 43 EA D A1 73 7D B7 68 D2 1C 16 75 52 1 0 DF C7 1 0 5E 0 0 0 59 0 0 0 B 0 0 0 FD FF FF FF 21 0 0 0 B 0 0 0 37 65 2A 1 F 0 0 0 14 97 2D 0 9F 0 0 0 12 19 26 3B 0 0 0 0 0 0 0 0

SparkFun Ublox Example
GPS: trying 38400 baud
GPS serial connected
Current update rate:1
Lat: 370987622 Long: -1216515166 (degrees * 10^-7) Alt: 86681 (mm) SIV: 13
Lat: 370987623 Long: -1216515166 (degrees * 10^-7) Alt: 86709 (mm) SIV: 13
Lat: 370987623 Long: -1216515167 (degrees * 10^-7) Alt: 86700 (mm) SIV: 13
Lat: 370987622 Long: -1216515167 (degrees * 10^-7) Alt: 86704 (mm) SIV: 12
Lat: 370987622 Long: -1216515167 (degrees * 10^-7) Alt: 86717 (mm) SIV: 12
Lat: 370987623 Long: -1216515166 (degrees * 10^-7) Alt: 86734 (mm) SIV: 12
Lat: 370987622 Long: -1216515167 (degrees * 10^-7) Alt: 86737 (mm) SIV: 12
Lat: 370987624 Long: -1216515168 (degrees * 10^-7) Alt: 86734 (mm) SIV: 12
Lat: 370987624 Long: -1216515168 (degrees * 10^-7) Alt: 86756 (mm) SIV: 12
Lat: 370987624 Long: -1216515168 (degrees * 10^-7) Alt: 86758 (mm) SIV: 12
Lat: 370987625 Long: -1216515168 (degrees * 10^-7) Alt: 86766 (mm) SIV: 12
```

### Actual behavior

It is always failing with the return code SFE_UBLOX_STATUS_CRC_FAIL in line 1160. If you comment out the line, the software works as expected. I understand the code is probably adapted primarily for I2C and the checksum may have some pecularities when used as such. Kindly request the UART and checksum work at the same level of functionality. Commenting out the checksum code seems to work fine after over 36hrs of use in a quite dynamic load testing environment.

PaulZC, 2020-04-16:
Hi James (@jafrado),
Thank you for the very comprehensive Issue report!
I am seeing the same issue on a Uno (ATmega328P) plus a NEO-M8T, using software serial.
Can you please tell me if: you installed the library through the library manager (and so are using version 1.7.2); or whether you downloaded a zip from GitHub - and if so can you remember what day you downloaded it on?
Many thanks,
Paul

jafrado, 2020-04-16:
@PaulZC - thank you for your response. I am using the latest TOT master library (copied into folder). Is there another released version I should use (either in source or library version)?
Thank you!

jafrado, 2020-04-16:
@PaulZC - thank you for the outstanding work here. Just tested (86e78de46cea8fbac9ad35de3d3f3ffccab1b877) and it works great! Closing this issue.

closed 2020-04-16 by jafrado

PaulZC, 2020-04-16:
Hi James (@jafrado),
Don't worry - I was just trying to understand if you were using the last release (downloaded through the Arduino Library Manager) or whether you had downloaded the latest code of the library direct from GitHub.
I have made several changes to the library over the past few hours (I'm on UK time). I wonder if you could download and test the library again please? I'm hoping all of your problems will vanish before your eyes.
No rush - whenever is convenient.
If you are still having problems, could you please post more debug messages?
Best wishes,
Paul

PaulZC, 2020-04-16:
Ah, our messages overlapped!
Thanks for letting me know.
All the best.

jafrado, 2020-04-20:
Thanks Paul,

No problem :-)
I have some changes to send you later for ECEF and NED. Will post a merge
request later this week if that is ok .

Best regards
-James