sparkfun/SparkFun_Ublox_Arduino_Library #91: SENDING RXM-PMREQ, timeout

engelberger5353, 2020-04-17:
Hello, I have tried to create the following function to send the zed f9p module to sleep, however I always receive a timeout, I have tried variations of the payload as well as disabled GLONASS and the result is the same TimeOut, unfortunately there is nothing on the web to know if it is a library or technical error

void SFE_UBLOX_GPS::setSLEEP(long slp, uint8_t uartPort, uint16_t maxWait)
{
  //Get the current config values for the UART port
  getPortSettings(uartPort, maxWait); //This will load the payloadCfg array with current port settings

 
  packetCfg.cls = UBX_CLASS_RXM;
  packetCfg.id = UBX_RXM_PMREQ;
  packetCfg.len = 16;
  //packetCfg.startingSpot = 0;

  //payloadCfg is now loaded with current bytes. Change only the ones we need to
  
  payloadCfg[0] = slp;
  payloadCfg[1] = slp >> 8;
  payloadCfg[2] = slp >> 16;
  payloadCfg[3] = slp >> 24;
  payloadCfg[4] = 0x00;
  payloadCfg[5] = 0x00;
  payloadCfg[6] = 0x00;
  payloadCfg[7] = 0x00;

  
  sfe_ublox_status_e retVal = sendCommand(packetCfg, 10000);  
  
  
  if (_printDebug == true)
  {
  _debugSerial->print("SLEEP OF:");
  _debugSerial->println(((uint32_t)payloadCfg[3] << 24) | ((uint32_t)payloadCfg[2] << 16) | payloadCfg[1] << 8 | payloadCfg[0]);
    
  _debugSerial->println(F("SLEEP: sendCommand returned: "));
  _debugSerial->println(statusString(retVal));
  }
  
}

can you help me?

nseidle, 2020-04-17:
Can you reference the section of the ZED-F9P datasheet that discusses power saving? I can't get access to them at the moment but if my memory serves me, the ZED doesn't support power saving.

PaulZC, 2020-04-18:
Hi @engelberger5353 ,
Looking at the ZED-F9P protocol description, there are two versions of the command:
[image]
and
[image]
[image]

I have not tried this, but looking at your code:
- First, please make sure you are adding this directly to the library .cpp and .h. Your code won't be able to see packetCfg unless it is in the library (packetCfg is private)
- Your call to getPortSettings seems unnecessary?
- slp needs to be uint32_t (not long) and .len needs to be 8 (not 16)

This might work:

```
// slp needs to be uint32_t (unsigned long)
packetCfg.cls = UBX_CLASS_RXM;
packetCfg.id = UBX_RXM_PMREQ;
packetCfg.len = 8;
packetCfg.startingSpot = 0;
payloadCfg[0] = slp;
payloadCfg[1] = slp >> 8;
payloadCfg[2] = slp >> 16;
payloadCfg[3] = slp >> 24;
payloadCfg[4] = 0x00; // Change this to 0x02 if you want to use backup mode
payloadCfg[5] = 0x00;
payloadCfg[6] = 0x00;
payloadCfg[7] = 0x00;
```
Best wishes,
Paul

PaulZC, 2020-04-24:
Hello @engelberger5353 ,
V1.8.0 of the library has a new custom packet feature which you can use to define your own packet for PMREQ from inside a normal .ino. Please see the new [Example20](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/master/examples/Example20_SendCustomCommand/Example20_SendCustomCommand.ino) for details.
I am going to close this issue but please reopen it if you are still having problems.
Best wishes,
Paul

closed 2020-04-24 by PaulZC