sparkfun/SparkFun_Ublox_Arduino_Library #100: Examples 1 and 2 fail to compile due to checkUblox() function call

hoilett, 2020-04-26:
### Subject of the issue
It appears some of the examples (at least Example1_BasicNMEARead and Example2_NMEAParsing) fail to compile (Arduino IDE version 1.8.12). The line in question is the myGPS.checkUblox() call. The checkUblox function appears to expect input parameters, but no parameters are passed in the example sketches. I don't have any hardware. I was simply trying to compile the code for an Arduino Uno. Library version 1.8.0 and MicroNMEA library version 2.0.1.

I haven't checked any of the other examples, but I imagine they would have the same problem.

rcal42, 2020-04-26:
I also got the same thing (I DO have hardware, but compile check does not need it).
Same thing happens in "Zed-F9P->Example3_StartRTCMBase"
..Documents/Arduino/libraries/SparkFun_Ublox_Arduino_Library/src/SparkFun_Ublox_Arduino_Library.h:465:10: note:   candidate expects 3 arguments, 0 provided
exit status 1
no matching function for call to 'SFE_UBLOX_GPS::checkUblox()'

Examples like Example3_Position work just fine - they compile and run on the hardware
There is a consistent issue with checkUblox()
I noticed some of the src was just updated 4 days ago.
I am not skilled enough to debug .cpp and .h
Great libraries, and great boards - looking forward to having it working

PaulZC, 2020-04-27:
Hi @hoilett & @rcal42,
Thank you for spotting this!
We changed the way checkUblox works in V1.8.0 but unfortunately it broke the backward-compatibility with the examples which call checkUblox directly.
This has been corrected in V1.8.1.
Tested on SparkFun RedBoard Qwiic (ATmega328P) with ZED-F9P using both I2C and SoftwareSerial:
- Example13_PVT\Example2_AutoPVT_ExplicitUpdate : pass
- Example13_PVT\Example3_AssumeAutoPVTviaUart : pass (autoPVT has to be enabled separately)
- Example1_BasicNMEARead : pass
- Example2_NMEAParsing : pass
- ZED-F9P\Example3_StartRTCMBase : pass (RTCM data seen)
- Example6_EnableNMEASentences : pass

Please let me know if V1.8.1 corrects your compilation issues - it should.
I'm confident this is fixed but I will leave this issue open for a few days, just in case anyone else has the same problem. 
Best wishes,
_**Paul**_

rcal42, 2020-04-27:
Confirmed fixed!
(Now that is service - fixed in less than 24 hrs - thanks!)
Compile works on the examples I tested before - Example1_BasicNMEARead, and "Zed-F9P->Example3_StartRTCMBase"
I built and ran Example1_BasicNMEARead successfully on Zed-F9P hardware.

closed 2020-05-01 by PaulZC