sparkfun/SparkFun_Ublox_Arduino_Library #102: I am trying to assess the accuracy of the SparkFun GPS-RTK-SMA Breakout - ZED-F9P (Qwiic)

zestpara, 2020-04-28:
### Subject of the issue
Describe your issue here. If you reference a datasheet please specify which one and in which section (ie, the protocol manual, section 5.1.2). Additionally, screenshots are easy to paste into github.

### Your workbench
* What development board or microcontroller are you using?
* What version of hardware or breakout board are you using? 
* How is the breakout board wired to your microcontroller?
* How is everything being powered?
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue.

### Expected behavior
Tell us what should happen

### Actual behavior
Tell us what happens instead

zestpara, 2020-04-28:
How do i find the closest RTK internet feed (RTCM) near Stouffville,Ontario,Canada ?

PaulZC, 2020-04-29:
Hi @zestpara ,
GitHub Issues are a way of reporting problems with library code. For questions like this, please use [the forum](https://forum.sparkfun.com/viewforum.php?f=116).
After a quick search for "NTRIP Canada", I found [this](https://www.nrcan.gc.ca/maps-tools-publications/tools/geodetic-reference-systems-tools/data/10923). Hopefully that answers your question.
Best wishes,
Paul

closed 2020-04-29 by PaulZC