sparkfun/SparkFun_Ublox_Arduino_Library #107: SerialUSB issue with non-SparkFun SAMD boards

adamgarbo, 2020-05-13:
### Subject of the issue
All code examples fail to compile when using an Adafruit M0 based microcontroller.

### Your workbench
* What development board or microcontroller are you using?
  * Adafruit Adalogger M0

### Steps to reproduce
* Compile code with Adafruit Feather M0 board definition selected.

### Expected behavior
* Code compiles as normal.

### Actual behavior
```
/Arduino/libraries/SparkFun_Ublox_Arduino_Library/src/SparkFun_Ublox_Arduino_Library.h:81:16: error: 'SerialUSB' was not declared in this scope
 #define Serial SerialUSB
```

### Solution
* Remove lines 79-82 introduced in v1.8.0.
```
// Define Serial for SparkFun SAMD based boards.
#if defined(ARDUINO_ARCH_SAMD)
#define Serial SerialUSB
#endif
```

PaulZC, 2020-05-14:
Hi Adam,
The problem is that most of the SparkFun SAMD boards default to SerialUSB. Serial is undefined on those boards so if we remove lines 79-82 we see compilation errors there instead.
Can you please try this branch and let me know if it fixes the problem?
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/tree/PaulZC__SAMD_SerialUSB_fix
Many thanks,
Paul

PaulZC, 2020-05-14:
The change I've made replaces lines 79-82 in the header file with:
```
// Define Serial for SparkFun SAMD based boards.
// Boards like the RedBoard Turbo use SerialUSB (not Serial).
// But other boards like the SAMD51 Thing Plus use Serial (not SerialUSB).
// The next nine lines let the code compile cleanly on as many SAMD boards as possible.
#if defined(ARDUINO_ARCH_SAMD) // Is this a SAMD board?
	#if defined(USB_VID) // Is the USB Vendor ID defined?
		#if (USB_VID == 0x1B4F) // Is this a SparkFun board?
			#if !defined(ARDUINO_SAMD51_THING_PLUS) // If it is not a SAMD51 Thing Plus
				#define Serial SerialUSB // Define Serial as SerialUSB
			#endif
		#endif
	#endif
#endif
```

adamgarbo, 2020-05-14:
Hi Paul,

This branch looks to compile properly for the Adafruit M0 Adalogger. Thanks for the quick fix!

Most users of SparkFun SAMD-based boards, including myself, have likely gotten into the habit of adding an extra #define Serial SerialUSB line at the top of their code. This is a nice way to avoid having to do so, but I'm wondering if it might be better suited to the actual board definitions, as opposed to a single library. I've never fully understood why this extra step is necessary for SparkFun SAMD boards, and not others. 

Cheers,
Adam

closed 2020-05-14 by adamgarbo

PaulZC, 2020-05-14:
Thanks Adam,
So, what's your vote?
- Merge the new change
- Leave the library as it is now
- Delete lines 79-82

Cheers,
Paul

adamgarbo, 2020-05-14:
Hi Paul,

I would vote to merge the new change if you're confident the nested #if statements will correctly target only SparkFun boards and set #define Serial SerialUSB.

Otherwise, I'd suggest deleting lines 79-82 and letting users of SparkFun SAMD-based boards carry on life as normal with manually needing to define SerialUSB (this is mentioned specifically in the Hookup Guides).

Cheers,
Adam