sparkfun/SparkFun_Ublox_Arduino_Library #108: Accuracy is not as reported in U-Center

biorove, 2020-05-17:
### Subject of the issue
Using a Zed-F9P Board with a Taoglas AGPSF.36C antenna. Running the Example 4 code for a base station with LCD `examples/ZED-F9P/Example4_BaseWithLCD/Example4_BaseWithLCD.ino` .
 I am getting an accuracy reported from the Arduino code around 10m, even after a few hours. However when I plug the board into U-Center it is saying the accuracy is 2.7 m. Is the accuracy being reported correctly in the Example 4 code? Why are the two reported values different. I would like an accuracy of less than 1m however the code runs forever stalling at  9-10m. 

### Your workbench
* What development board or microcontroller are you using? - Arduino Uno
* What version of hardware or breakout board are you using?  - Zed F9p GPS-RTK2
* How is the breakout board wired to your microcontroller? - QWICC onnector
* How is everything being powered? - Cable from PC to Arduino

PaulZC, 2020-05-18:
Hi @biorove ,
Thank you for raising this issue. I am investigating it for you.
Unfortunately I do not have an LCD here, so I am trying to replicate the issue with Example3_StartRTCMBase.
Can you please try two things for me?
Using Example4 as you have been:

Can you please open the Serial Monitor in the Arduino IDE while the code is running. When the Survey has started, you should see Serial messages containing the "Time elapsed" and "Accuracy". Do these match what is displayed on the LCD? (I'm trying to work out if this is actually a problem with the way the LCD prints floating point numbers. It seems unlikely, but you never know!)

If the Serial messages match what you see on the LCD, then please try this:
Please follow this link:
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/tree/correcting_svin
This will take you to a new "branch" of the library where I am trying to correct your issue.
Click on the green "Clone or download" button. Click "Download Zip".
This will download a copy of my changes in a Zip file.
Open the Zip file and navigate to the ```SparkFun_Ublox_Arduino_Library-correcting_svin\src``` folder.
In there you will see two files. Please copy the file called ```SparkFun_Ublox_Arduino_Library.cpp``` and paste it into your ```Arduino\libraries\SparkFun_Ublox_Arduino_Library\src folder```. (You might want to rename the original .cpp to something like ```SparkFun_Ublox_Arduino_Library.cpp.old``` first so you can easily restore it afterwards if you want to.) Upload Example4 to your Arduino as normal - the IDE will pick up the new version of the library when it compiles the code.
Does that correct the issue?

Here is what I see using Example3 on a RedBoard Qwiic (Arduino Uno) with a ZED-F9P and a Taoglas A.80.A.101111 antenna. I see an accuracy of 1.2m in both the Serial Monitor and in u-center.

[image]

Very best wishes,
Paul

PaulZC, 2020-05-18:
Hi @biorove ,
One more thought on this:
When you say "the accuracy is 1.7 m" in U-Center, what accuracy are you referring to?
Are you looking at the UBX-NAV-SVIN Mean 3D StdDev accuracy? Or maybe the HDOP or PDOP?
HDOP and PDOP will be much lower than the UBX-NAV-SVIN Mean 3D StdDev accuracy:

[image]

Best wishes,
Paul

biorove, 2020-05-19:
Thanks Paul for the help!
 Yes the serial monitor and the LCD show the same messages. 
I believe my confusion on the accuracy was that I was looking at the 
UBX-NAV (Navigation) HPPOSECEF (High Precision Position ECEF) instead of the SVIN! When I look at the SVIN accuracy it is the same as what is being displayed on the LCD and Serial Monitor. What is the difference between the two?

I tried both the original library and your new one and both work! However I do have another question on the library and the time elapsed/ survey in progress. 

Today I set the module outside and left it for about 1:30 hrs and was getting a survey in accuracy of 0.40 m. (I set the survey in at 60 seconds / / 0.100 m)  Since I set the code to 0.100 m, the survey was not completed yet. I pulled the power from the Arduino and everything shut down. However when I powered it back up in a completely different location the LCD and Serial Monitor said "Survey in Progress" with an elapsed time of 5000ish seconds and still read 0.45. Like it was picking up where it left off. I left it for a few minutes and it went up to 0.50 m.  I even validated this in U-Center SVIN. It only changed when I uploaded Example4 again. 

Why when the module is powered down and then started back up does the survey not start over and elapsed time set back to zero? The same result happens when I try uploading a different version of Example 4 with a different survey in target accuracy. Say I upload a new version of Example 4 with a target accuracy of 3.000, the time elapsed does not reset.

In the serial monitor I even sent an "x" to stop the survey to transmit RTCM then pulled the power. When I power the arudino back up it starts at the previous elapsed time (both in serial monitor and lcd).

 

Thanks!!

PaulZC, 2020-05-19:
Hi @biorove ,
You're very welcome.
The GPS forum is the best place to ask questions about how to get the best out of the GNSS products. We use the GitHub issues to address problems with the code. But don't worry, your issue certainly did look like a problem with the code!

https://forum.sparkfun.com/viewforum.php?f=116

u-blox have some excellent white papers that describe how the different aspects of GNSS positioning work. These should answer your questions:

https://www.u-blox.com/en/product-resources/field_file_category/white-paper-212/field_file_products%253Afield_product_category/position-time-152

https://www.u-blox.com/en/ubx-viewer/view/GPS-Compendium_Book_(GPS-X-02007)?url=https%3A%2F%2Fwww.u-blox.com%2Fsites%2Fdefault%2Ffiles%2Fproducts%2Fdocuments%2FGPS-Compendium_Book_%2528GPS-X-02007%2529.pdf

https://www.u-blox.com/en/white-paper-design-guide-high-performance-gnss-patch-antenna

https://www.u-blox.com/en/node/14861

Regarding the module remembering that a survey is in progress:
Yes, the example will do that. It's all down to [this line of code](
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/master/examples/ZED-F9P/Example4_BaseWithLCD/Example4_BaseWithLCD.ino#L72):
```
myGPS.saveConfiguration(); //Save the current settings to flash and BBR
```
We use that line to tell the module to remember the I2C port settings and that NMEA is disabled (set by the previous line). If you run the code more than once without removing the power, the survey-in also gets saved in battery-backed memory.
You can stop this by commenting that line (add two slashes in front):
```
//myGPS.saveConfiguration();
```
To reset the module back to its initial state, add the following line just after the [Wire.setClock](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/master/examples/ZED-F9P/Example4_BaseWithLCD/Example4_BaseWithLCD.ino#L65):
```
myGPS.factoryDefault();
```
You need to do it before the [myGPS.setI2COutput](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/master/examples/ZED-F9P/Example4_BaseWithLCD/Example4_BaseWithLCD.ino#L71) because resetting to factoryDefault re-enables the NMEA messages.
I'm going to close this issue as we have ruled out a problem with the code, but please feel free to reopen it if required.
Very best wishes,
Paul

closed 2020-05-19 by PaulZC