sparkfun/SparkFun_Ublox_Arduino_Library #117: Example3 ZED F9P no longer works after firmware update 1.13

ramsie838, 2020-08-03:
### Subject of the issue
After installing the update  ZED-F9P HPG 1.13 firmware the example Example3_StartRTCMBase.ino no longer works. There is a problem with achieving the base survey. The serial monitor only outputs "failed to get survey status".

Here is the code snippet:

  //Check if Survey is in Progress before initiating one
  response = myGPS.getSurveyStatus(2000); //Query module for SVIN status with 2000ms timeout (request can take a long time)
  if (response == false)
  {
    Serial.println("Failed to get Survey In status");
    while (1); //Freeze
  }

### Your workbench
* What development board or microcontroller are you using?
Arduino Uno Rev. 3

GPS RTK2 ZED-F9P

* How is the breakout board wired to your microcontroller?
Connected via a qwiic cable.
GND->GND
3,3V->3,3V
SDA->SDA
SCL->SCL

* How is everything being powered?
Over Arduino USB

* Are there any additional details that may help us help you?

### Steps to reproduce
Connect your Arduino R3 with the Sparkfun GPS RTK2 ZED-F9P via QWIIC Cable.

Connections:
GND->GND
3,3V->3,3V
SDA->SDA
SCL->SCL

Upload Example3_StartRTCMBase.ino. Open the serial monitor, press a key and see what happens.

### Expected behavior
serial monitor prints: Base survey complete! RTCM now broadcasting.

## Actual behavior
serial monitor prints: Survey start failed.

If you upload Example3_StartRTCMBase.ino to the arduino and then disconnect the PC and the GPS RTK2 ZED-F9P. And then do a firmware update 1.13 via ucenter, it will work if you do not upload the code from Example3_StartRTCMBase.ino to the arduino again.

PaulZC, 2020-08-03:
Thanks @ramsie838 - we will investigate this.
It looks like U-blox have made some changes to the RTCM functionality. Please see section 4 [in the release notes.](https://www.u-blox.com/en/docs/UBX-20019211)
Best wishes,
Paul

PaulZC, 2020-08-04:
Hi @ramsie838 ,
I can not replicate the problem you are having. I have upgraded my ZED-F9P to 1.13 and the example still works OK for me.
I do not have an Arduino Uno but I have tried the code on a SparkFun RedBoard Qwiic (ATmega328P) and it works OK.
It works OK on the RedBoard Artemis ATP too.
Some suggestions:
Please make sure you are using the latest version of the u-blox library. Please upgrade if you need to.
Maybe the RTCM messages are causing the ```getSurveyStatus``` to time out, if you have run the example more than once?
Can you please try adding these lines:
```
  if (myGPS.factoryDefault() == false)
  {
    Serial.println("***!!! Warning: factoryDefault failed !!!***");
  }
  else
  {
    Serial.println("Configuration set to factory default!");
  }
```
before the
```
  myGPS.setI2COutput(COM_TYPE_UBX); //Set the I2C port to output UBX only (turn off NMEA noise)
  myGPS.saveConfiguration(); //Save the current settings to flash and BBR
```
This will make sure the RTCM messages are disabled for the ```getSurveyStatus```.
Please let us know if this fixes your problem.
Many thanks,
Paul

ramsie838, 2020-08-04:
After adding these lines:

/*  if (myGPS.factoryDefault() == false)
  {
    Serial.println("***!!! Warning: factoryDefault failed !!!***");
  }
  else
  {
    Serial.println("Configuration set to factory default!");
  }

before the

  myGPS.setI2COutput(COM_TYPE_UBX); //Set the I2C port to output UBX only (turn off NMEA noise)
  myGPS.saveConfiguration(); //Save the current settings to flash and BBR*/

it works again.

Serial monitor output:

10:11:21.392 -> Ublox Base station exation set to factory default!
10:11:21.392 -> Press any key to send commands to begin Survey-In
10:11:21.392 -> Ublox Base station example
10:11:21.631 -> Configuration set to factory default!
10:11:21.941 -> Press any key to send commands to begin Survey-In
10:11:26.405 -> RTCM messages enabled
10:11:26.715 -> Survey started. This will run until 60s has passed and less than 5m accuracy is achieved.
10:11:27.508 -> Press x to end survey - Time elapsed: 0 Accuracy: 2.56
10:11:29.535 -> Press x to end survey - Time elapsed: 2 Accuracy: 2.52
10:11:31.535 -> Press x to end survey - Time elapsed: 4 Accuracy: 2.48 
...

I have done another firmware update and it works fine again!

Many thanks Paul.

PaulZC, 2020-08-04:
Hi @ramsie838 ,
Great - I'm glad that solved the problem. (I was going to suggest changing the I2C pull-ups next, but I'm glad we did not need to go there!)
```saveConfiguration``` causes the ZED to store all of its settings in BBR. The first time you run the example, the only change that gets saved is the I2C port settings. But running the code a second time causes the RTCM settings to be saved too. The RTCM messages then cause a lot of additional I2C traffic which, I think, caused the ```getSurveyStatus``` to time out. If you are writing your own code, you can use ```saveConfigSelective``` instead to only save some of the settings.
Best wishes,
Paul

closed 2020-08-04 by PaulZC