sparkfun/SparkFun_Ublox_Arduino_Library #120: esp32 and m8u dead reckoning not working

joeholdsworth, 2020-08-11:
### Subject of the issue
in i2c mode the m8u dead reckoning never calibrates, if I use it in uart mode with ublox center it does. Some I2C requests occasionally do not work. 

### Your workbench
* What development board or microcontroller are you using - ESP32
* What version of hardware or breakout board are you using - sparkfun neo m8u
* How is the breakout board wired to your microcontroller - I2C lines, 3.3v supply capable of 300mA
* How is everything being powered - via an LDO 3.3v 300mA reg
* Are there any additional details that may help us help you - id like to perform a check that the DR mode is enabled, is there a command for this?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue.

I'm using the sparkfun examples provided, specifically the DR fusion calibration status,  it never achieves calibration,if I perform the same vehicle maneuvers in uart mode and use the esp32 as a serial to usb bridge with the ublox center, it does.

### Expected behavior
fusion calibration staus should indicate calibrated

### Actual behavior
unit never calibrates

Daedaliss10, 2020-08-14:
I am also having an issue developing on a ESP board using this library.  

### Subject of the issue
NEO-M8U module causes bus issues with other I2C devices, causes resets and general instability.

Your workbench
What development board or microcontroller are you using - ESP-12F ( Wemos D1 Mini)
What version of hardware or breakout board are you using - Sparkfun GPS Board (NEO-M8U)
How is the breakout board wired to your microcontroller - I2C lines, 3.3v supply capable of 300mA
How is everything being powered - via an LDO 3.3v 300mA reg
Are there any additional details that may help us help you - Will be using 3 devices on I2C bus.

### Steps to reproduce
### SETUP:
Install 3 I2C devices on Wemos D1 Mini; Sparkfun NEO-M8U, Sparkfun MAX30105 & Sparkfun BME680.  Install libraries to connect and communicate with the devices.  Remove jumper connections for I2C bus pull ups on two of three devices.  Set up basic sketch to connect and poll sensors.  Connect Serial monitor for output messages.   (sketch to follow)
### ISSUE:
Device behavior is erratic. Sometimes sketch gets through to the main loop, sometimes gets stopped in failed device connects in during setup.  (sometimes fresh boot works, resetting device via RESET pin does not).  Any messages from GPS module are garbled and not consistent.  

### Sketch: 
#include "SparkFun_Ublox_Arduino_Library.h" //http://librarymanager/All#SparkFun_Ublox_GPS

#include "heartRate.h"
#include <Wire.h>
#include "MAX30105.h"

#include "spo2_algorithm.h"
#include "Zanshin_BME680.h"

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <EEPROM.h>

BME680_Class BME680; ///< Create an instance of the BME680 class
//BME60 I2C address = 0x77 (0x76 if jumper closed on bottom)

MAX30105 particleSensor;
//MAX30105 I2C address = 0xAF (read), 0xAE (write)?  .h file says 0x57

SFE_UBLOX_GPS myGPS;
//GPS I2C address = 0x42

static int32_t  temp, humidity, pressure, gas;
void setup() {

  pinMode(16, OUTPUT);
  pinMode(A0, INPUT);
  digitalWrite(16, LOW);
  // put your setup code here, to run once:
  Serial.begin(115200); // Start serial port at Baud rate
  Wire.begin();

  while (!BME680.begin(I2C_STANDARD_MODE)) // Start BME680 using I2C, use first device found
  {
    Serial.print(F("-  Unable to find BME680. Trying again in 5 seconds.\n"));
    delay(1000);
  }

  while(particleSensor.begin() == false)
  {
    Serial.println("MAX30105 was not found. Please check wiring/power. ");
    delay(1000);
  }

  particleSensor.setup(); //Configure sensor. Use 6.4mA for LED drive

  while(myGPS.begin() == false) //Connect to the Ublox module using Wire port
  {
    Serial.println(F("Ublox GPS not detected at default I2C address"));
    delay(1000);
  }

  myGPS.setI2COutput(COM_TYPE_UBX); //Set the I2C port to output UBX only (turn off NMEA noise)
  //myGPS.setNMEAOutputPort(Serial);

  BME680.setOversampling(TemperatureSensor,Oversample16); // Use enumerated type values
  BME680.setOversampling(HumiditySensor,   Oversample16); // Use enumerated type values
  BME680.setOversampling(PressureSensor,   Oversample16); // Use enumerated type values
  BME680.setIIRFilter(IIR4); // Use enumerated type values
  BME680.setGas(320,150); //  for 150 milliseconds
}

void loop() {
  temp = 0;
  humidity = 0;
  pressure = 0;
  gas = 0;
  BME680.getSensorData(temp,humidity,pressure,gas);
  String BMEoutput = "BME680 output: temp = " + String(temp) + ", humidity = " + String(humidity) + ", pressure = " + String(pressure) + ", gas + " + String(gas);
  Serial.println(BMEoutput);
  Serial.println("BME.I2CAddress = " + String(BME680.getI2CAddress()));
  Serial.println("MAX30105 R = " + String(particleSensor.getRed()) + ", G = " + String(particleSensor.getGreen()) + ", IR = " + String(particleSensor.getIR()));
  Serial.println("u-blox.IsConnected() = " + String(myGPS.isConnected(100)));
  //Serial.println("u-blox version = " + String(myGPS.getProtocolVersionHigh()) + "." + String(myGPS.getProtocolVersionLow()));
  //myGPS.checkUblox(); //See if new data is available. Process bytes as they come in.
  Serial.println("Analog Reading = " + String(analogRead(A0)));
  digitalWrite(16, !digitalRead(16));
  delay(5000);
}

### 2ND TEST
### SETUP:
Removed all but the the GPS module from I2C bus on Wemos D1 Mini. Run Example3_GetPosition from Sparkfun u-blox Arduino library. No I2C pull up resistors on GPS module OR Wemos D1 Mini (as per Wemos schematic)

### ISSUE:
Similiar behavior as before.  Sometimes resets once or two on startup, then gets into loop and correctly functions.  On system reset, never re-establishes connection and continues to reset.

### 3RD TEST
Same test as 2ND TEST, this time including a I2C with pull ups, but not configured in firmware.  Run Example3_GetPosition again. Similar behavior as 2ND TEST.

### Expected behavior
Stable and reliable device operation when used on I2C bus with other devices installed.

### Actual behavior
System operation is unstable and unreliable.  Not usable in current form.

PaulZC, 2020-10-05:
Hi @joeholdsworth ,

Apologies for the very slow reply - this fell off my radar...

Are you having the same issue as [this one](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/issues/121#issuecomment-703634605)?

Best wishes,

Paul

PaulZC, 2020-11-30:
Hi @joeholdsworth ,
Thank you for your patience!
It has taken a while, but I think we have found the root cause of this issue. There was a gremlin in the code that meant the calibration status wasn't being extracted correctly. The correction is here:
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/commit/04d8f4a2f70590d73335247d664f50282246ff12#diff-56e995c8ccf5d6b3bc269a8eaf038164594becf0f1c2df99616a0cd4f803c028R4098
The changes are still in the release_candidate branch, but I hope to be able to merge them into a new release in a day or two.
Thanks again for reporting this.
Best wishes,
Paul

closed 2020-12-05 by PaulZC