sparkfun/SparkFun_Ublox_Arduino_Library #122: Trouble uploading Example19_DynamicModel sketch to atmega2560

galagaffe, 2020-09-18:
### Subject of the issue
Sending Example19_DynamicModel sketch to an atmega2560 fails. The other example sketches i tested   out of the repository are working fine. 

### Your workbench
* What development board or microcontroller are you using?
ATMEGA2560
* What version of hardware or breakout board are you using? 
MEGA Sensor Shield v2

SAM M8Q
Hard version: 00080000
Extensions:4
  FWVER=SPG 3.01
  PROTVER=18.00
  GPS;GLO;GAL;BDS
  SBAS;IMES;QZSS

* How is the breakout board wired to your microcontroller?
I2C 
* How is everything being powered?
5v USB Bus Power from the shield converted with a ldo regulator AMS1117 to 3.3V
* Are there any additional details that may help us help you?
This behavior occurs when the function setDynamicModel or getDynamicModel is used. 

### Steps to reproduce
Just send the sketch using the Arduino IDE 1.8.3 and the library version 1.8.6 installed with the library manager

### Expected behavior
sketch should be sent to the board and starts working

### Actual behavior
Arduino IDE status uploading RX TX leds are blinking from time to time. After a while Arduino IDE returns "an error occured while uploading the sketch".

PaulZC, 2020-09-18:
Hi @galagaffe ,
So, to be clear, the other examples are working on your ATmega2560? Have you tried one of the other examples since trying Example19? Are you sure the other examples still work? (I'm wondering if your board(s) have developed a fault?)
Also, what voltage are you using for I2C? Have you reduced the I2C voltage to 3.3V too - using a level-shifting circuit?
Which SAM-M8Q breakout board are you using? (Is it a SparkFun one?)
Best wishes,
Paul

galagaffe, 2020-09-19:
Good morning Paul,thanks a lot for replying. Yes the other examples are working fine. I guess like you that there must be a glitch on the board because the communication to the board get lost at the precise moment the first byte of the compiled hex file is sent.It's just strange that only happens using this function.I'm using a AMS1117 low drop regulator to convert from 5v to 3.3V. This is the gps breackout board I use.
CAM-M8Q-Breakout Multi GNSS Modul (GPS, QZSS, GLONASS, BeiDou, Galile

| 
| 
| 
| €25.44 |  |

 |

 |
| 
|  | 
CAM-M8Q-Breakout Multi GNSS Modul (GPS, QZSS, GLONASS, BeiDou, Galile

Breakout Board mit dem u-blox CAM-M8Q Multi GNSS Modul zum Empfang von GPS, QZSS, GLONASS, BeiDou und Galileo Daten
 |

 |

 |

But as you said as other sketches are working fine it's probably my atmega board. I will get another one and do a cross check.Thanks for you help.
Best regards from KarlsruheCédric

PaulZC, 2020-09-19:
Hi Cédric (@galagaffe ),
Thank you for the extra information. This is an interesting problem!
I have an ATmega2560 board here and Example19 compiles and uploads OK for me.
(I cannot test the code as I do not have a 5V-compliant u-blox board here. All of my u-blox boards are 3.3V only. But it does upload OK.)
I do not think I can offer you any more help, but I will leave this issue open for a while - just in case I think of anything.
Attached below are the messages I see using Arduino 1.8.**13**:
Please check your I2C signals. The CAM-M8Q breakout I2C uses 3.3V I2C. Your ATmega will be pulling up SCL and SDA to 5V through 10K resistors.
I hope you manage to solve this!
Very best wishes,
Paul

```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\pc235\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\pc235\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\pc235\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=2341_0042 -ide-version=10813 -build-path C:\Users\pc235\AppData\Local\Temp\arduino_build_92813 -warnings=none -build-cache C:\Users\pc235\AppData\Local\Temp\arduino_cache_108484 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\pc235\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\pc235\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=C:\Users\pc235\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\pc235\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=C:\Users\pc235\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\pc235\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\Users\pc235\Documents\Arduino\libraries\SparkFun_Ublox_Arduino_Library\examples\Example19_DynamicModel\Example19_DynamicModel.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\pc235\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\pc235\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\pc235\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=2341_0042 -ide-version=10813 -build-path C:\Users\pc235\AppData\Local\Temp\arduino_build_92813 -warnings=none -build-cache C:\Users\pc235\AppData\Local\Temp\arduino_cache_108484 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\pc235\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\pc235\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=C:\Users\pc235\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\pc235\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=C:\Users\pc235\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\pc235\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\Users\pc235\Documents\Arduino\libraries\SparkFun_Ublox_Arduino_Library\examples\Example19_DynamicModel\Example19_DynamicModel.ino
Using board 'mega' from platform in folder: C:\Users\pc235\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3
Using core 'arduino' from platform in folder: C:\Users\pc235\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3
Detecting libraries used...
"C:\\Users\\pc235\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\pc235\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\cores\\arduino" "-IC:\\Users\\pc235\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.3\\variants\\mega" "C:\\Users\\pc235\\AppData\\Local\\Temp\\arduino_build_92813\\sketch\\Example19_DynamicModel.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
"C:\\Users\\pc235\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\pc235\\AppData\\Local\\Arduino15\\packages\\ [... cut ...]
avrdude: verifying flash memory against C:\Users\pc235\AppData\Local\Temp\arduino_build_92813/Example19_DynamicModel.ino.hex:
avrdude: load data flash data from input file C:\Users\pc235\AppData\Local\Temp\arduino_build_92813/Example19_DynamicModel.ino.hex:
avrdude: input file C:\Users\pc235\AppData\Local\Temp\arduino_build_92813/Example19_DynamicModel.ino.hex contains 18038 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 2.33s

avrdude: verifying ...
avrdude: 18038 bytes of flash verified

avrdude done.  Thank you.
```

PaulZC, 2020-09-19:
Hi again Cédric (@galagaffe ),
I figured out how to make an I2C level-shifter with the components I had on-hand. The example works OK for me (see below).
Please check your I2C voltages. And please consider upgrading to Arduino 1.8.13.
Good luck!
Best wishes,
Paul
```
SparkFun Ublox Example
Dynamic platform model changed successfully!
The new dynamic model is: 0
```

closed 2020-09-19 by PaulZC