sparkfun/SparkFun_Ublox_Arduino_Library #123: UBX: counter hit MAX_PAYLOAD_SIZE

eried, 2020-10-01:
### Subject of the issue
I am getting that debug output message about UBX: counter hit MAX_PAYLOAD_SIZE, what is the reason behind that message?

### Your workbench
Artemis Global Tracker with ZOE-M8Q 

### Expected behavior
Silent operation

### Actual behavior
UBX: counter hit MAX_PAYLOAD_SIZE messages while waiting for SIV > 8

The piece of code doing this:
```

      if (verbose)
        myGPS.enableDebugging(); // Enable debug messages
        
      myGPS.setI2COutput(COM_TYPE_UBX); // Limit I2C output to UBX (disable the NMEA noise)

      long abort = millis() + GPS_ABORT_AFTER_MS;
      while (millis() < abort)
      {
        dispatcher.delay(3000);
        byte fixType = myGPS.getFixType();

        String d = "Waiting for FIX>0: ";
        d += fixType;
        debugln(d.c_str());

        if (fixType > 0)
        {
          byte SIV = myGPS.getSIV();

          String d = "Waiting for SIV>: ";
          d += GPS_MIN_SIV + ": ";
          d += SIV;
          debugln(d.c_str());

          if (SIV > GPS_MIN_SIV)
          {
            debugln("Waiting for valid time and date");

            if (myGPS.getTimeValid() && myGPS.getDateValid())
            {
```

I am just querying the values every 3 seconds, so I do not think it might be the source of problem

Thanks

PaulZC, 2020-10-02:
Hi Erwin (@eried ),
It means the payload array is full. And that the data receive lost synchronization in some way.
This normally happens if you have accidentally left the NMEA messages enabled, But I see that you are calling ```myGPS.setI2COutput(COM_TYPE_UBX);``` so it should not be that issue.
What navigation rate are you using? If you only need a fix every 3 seconds, then changing the navigation frequency may prevent this.
Please see Example7:
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/master/examples/Example7_OutputRate/Example7_OutputRate.ino
Also, please make sure you are using 100kHz clocking on the I2C. Running u-blox modules at 400kHz can cause issues.
Best wishes,
Paul

eried, 2020-10-02:
Hey, thanks. With those 2 changes works smoothly. I was just following the examples in the Artemis Tracker repo :)

```
myGPS.setNavigationFrequency(1); //Set output to 10 times a second
Wire1.setClock(100000);
```

eried, 2020-10-05:
Strange, with the frequency on 1 and clock 100000 I still get sometimes the `essUBX: counter hit MAX_PAYLOAD_SIZE`, less frequently, but still. What else could I try?

PaulZC, 2020-10-05:
Hi Erwin,

The next thing to try is to disable the I2C pull-ups on the Artemis pins (internal to the Artemis).

Please try adding this line of code before the **myGPS.begin**:
```Wire1.setPullups(0); //Disable pullups for u-blox comms```

You should see another reduction in the number of I2C errors when talking to the u-blox.

But, the MS8607 pressure sensor may stop working. It may need the pull-ups to be active. If that happens, please add
```Wire1.setPullups(1); //Enable 1.5k I2C pull-ups - internal to the Artemis```
before calling any of the ```barometricSensor``` functions.

Best wishes,

Paul

eried, 2020-10-05:
Thanks, is the same. But for now I will just keep debugging disabled while waiting for fix, because it actually works ok.

PaulZC, 2020-11-30:
Hi Erwin (@eried ),
How's it going? Are you still seeing these counter hit MAX_PAYLOAD_SIZE messages? How often?
Best wishes,
Paul

eried, 2020-11-30:
I discovered that I stupidly sent my working iridium modem instead of the non working artemis global tracker to sparkfun for RMA more than a month ago... so actually no more errors because I am using another platform now 😂

closed 2020-11-30 by eried