sparkfun/SparkFun_Ublox_Arduino_Library #126: Is there a way to read current chip settings?

dotMorten, 2020-10-16:
### Subject of the issue
There's a lot of commands to configure the u-blox chip, but there seems to be a lack of methods for getting the current configuration.
Example: I can set the port output settings like `setUART1Output`, but I can't get the current comSessions for UART1 (I do see a `getPortSettings` method but that returns a boolean and not `uint8_t`).
Similarly I can enable and disable NMEA and RTCM messages, but there's no way to get what the current setting is.

PaulZC, 2020-10-16:
Hi Morten (@dotMorten ),

```getPortSettings``` is used internally to read the current port settings. They are loaded into ```payloadCfg``` so they can be modified and pushed again using (e.g.) ```setPortOutput```. Yes, correct, there is no simple function call to read and return the actual settings.

However, you can read any of the settings using a custom command. Please have a look at Example20 (and Example21). You can set up a custom command for the setting you want to read, and then extract the setting from the returned customPayload. You do need to be familiar with the u-blox protocol specification, but this is very well documented:
- https://www.u-blox.com/en/docs/UBX-13003221
- https://www.u-blox.com/en/docs/UBX-18010854

If there is a setting you want to read regularly, and if you have the time, please go ahead and write a function to read it and send us a pull request. We'll be happy to merge it.

Best wishes,

Paul

closed 2020-10-16 by PaulZC

nseidle, 2020-10-23:
Hi @dotMorten - I too needed to be able to check a port config, then modify it only if the setting where not what I wanted. You can do it with a custom packet (see [example 20](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/master/examples/Example20_SendCustomCommand/Example20_SendCustomCommand.ino)) but it gets a bit messy. I am currently working on expanding the getVal/setVal methods. This makes config a bit easier and a bit more future proof. You can checkout the first example over on [this branch](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/Expand-getVal-setVal/examples/GetSetVal/Example1_GetSetPortSettings/Example1_GetSetPortSettings.ino).

dotMorten, 2020-10-23:
@nseidle Thanks! I pulled your changes over to try them out. First issue is I need an include guard at the top, but otherwise compiled fine.

Second, I see you have `UBLOX_CFG_UART1_BAUDRATE` and `UBLOX_CFG_UART2_BAUDRATE`, but I'm missing one for USB. How did you obtain those values?