sparkfun/SparkFun_Ublox_Arduino_Library #127: getHDOP and getVDOP missing

dotMorten, 2020-10-16:
### `getHDOP` and `getVDOP` missing

There is a method for getting the 3D dilution of precision but methods for getting the horizontal HDOP and vertical VDOP is missing. How do I retrieve these?

PaulZC, 2020-10-16:
Hi Morten (@dotMorten ),

We use the UBX-NAV-PVT message to read the lat, lon, time, and pDOP. See section 32.17.15 in the M8 protocol specification. Sadly that message does not contain HDOP and VDOP.

The solution is the same to your other question. You can set up a custom command to poll UBX-NAV-DOP (section 32.17.5 in the M8 specification) and then extract all of the DOP's from that.

Again, if you have the time, please do add getHDOP and getVDOP to the library and send us a pull request. We'll be happy to merge it.

Best wishes,

Paul

closed 2020-10-16 by PaulZC

dotMorten, 2020-10-16:
ok, I'll if I can figure that out. However should this issue be closed until then? I assume the PR I do make would be linked to this.

PaulZC, 2020-10-16:
Hi Morten,
The PR will get its own reference number. No problem there.
Have a great weekend,
Paul