sparkfun/SparkFun_Ublox_Arduino_Library #143: Different layer defaults when using newCfgValSet instead of setVal

markuckermann, 2020-10-29:
### Subject of the issue
Just a heads up really, I'v been caught out by the defaults for `newCfgValset...()` being different to `setVal()`. It's just `VAL_LAYER_BBR` instead of `VAL_LAYER_ALL` so the settings don't take effect. 

The defaults were recently changed in #134 and are set here: https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/97bd455b0e7a05b92bfe5a528c28cc47f2e2d289/src/SparkFun_Ublox_Arduino_Library.h#L577-L579

Is this on purpose? I'd suggest having the same defaults across the board. 

### Your workbench
* Teensy 3.6 and SFE GPS-RTK2 but it shouldn't matter.

### Steps to reproduce
```
myGPS.newCfgValset8(CFG_MSGOUT_RTCM_3X_TYPE1005_I2C, 1); 
myGPS.addCfgValset8(CFG_MSGOUT_RTCM_3X_TYPE1097_I2C, 1);
myGPS.sendCfgValset8(CFG_MSGOUT_RTCM_3X_TYPE1230_I2C, 10); 
```
### Expected behavior
Changes saved to RAM + other layers

### Actual behavior
Changes saved to BBR only

PaulZC, 2020-10-30:
Hi Mark (@markuckermann ),

Thanks for spotting this! I'll change it...
I'll leave this issue open until the fix is in...

Meanwhile, you can work around this by using:
```myGPS.newCfgValset8(CFG_MSGOUT_RTCM_3X_TYPE1005_I2C, 1, VAL_LAYER_ALL);```
etc.

Best wishes,

Paul

PaulZC, 2020-10-30:
The fix is in... On the release_candidate branch. I'll leave this open until it gets merged into master.
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/commit/0ef7394e57b7f7d5bd3938f228573b0c5c77ffd6

markuckermann, 2020-10-30:
Just realised it's explained in the example, but not very well.

https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/57c774df0e8891c64d91f2c4b0068af88646586b/examples/ZED-F9P/Example9_multiSetVal/Example9_multiSetVal.ino#L71

The comment suggests that you need to change the layer to have it updated in RAM. 

So as part of this fix, the example ought to be slightly updated too. @PaulZC

PaulZC, 2020-10-30:
OK smartypants! ;-D
Thanks Mark. Corrected here:
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/commit/5edcbf6e9caac52406a865d44636d2ca63b64b71#diff-f03994c30c299a3b99dffd9c6a5fd91df9fc47f7833a90e73a4f47f425a060afR71

closed 2020-12-05 by PaulZC