sparkfun/SparkFun_Ublox_Arduino_Library #148: Use of logical '&&' with constant operand

rubienr, 2020-11-16:
Multiple usages of logical `&&` with constant operand for example `imuMeas.dataType[i] = (bitField && 0xFF000000) >> 23` in:

https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/97bd455b0e7a05b92bfe5a528c28cc47f2e2d289/src/SparkFun_Ublox_Arduino_Library.cpp#L3597

https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/97bd455b0e7a05b92bfe5a528c28cc47f2e2d289/src/SparkFun_Ublox_Arduino_Library.cpp#L3632

https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/97bd455b0e7a05b92bfe5a528c28cc47f2e2d289/src/SparkFun_Ublox_Arduino_Library.cpp#L3646

https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/97bd455b0e7a05b92bfe5a528c28cc47f2e2d289/src/SparkFun_Ublox_Arduino_Library.cpp#L3670

https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/97bd455b0e7a05b92bfe5a528c28cc47f2e2d289/src/SparkFun_Ublox_Arduino_Library.cpp#L3706

PaulZC, 2020-11-17:
Many thanks @rubienr for spotting this! Yes, they should be bitwise ANDs (```&```).
@edspark - just copying you in so you are aware. I'll change this in the release_candidate branch ready for the next release.
Thanks again,
Paul

PaulZC, 2020-11-17:
Hi @rubienr & @edspark ,
@rubienr : thank you again for spotting this!
I have corrected the ANDs (and some of the bit masks) in [this commit](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/commit/04d8f4a2f70590d73335247d664f50282246ff12).
@edspark : can you please review the changes - and test if possible? I don't have a dead-reckoning module here to test them on...   :-(   Thanks!
Best wishes,
Paul

rubienr, 2020-11-17:
Hi,

I am at a very early stage with my project development and my prototype is stationary for the moment. Thus I cannot meningfully test dead-reckoning. However I switched my dependency to the RC branch. Will keep you posted if I find issues related to this commit. 

I tried to compile with `-Wall -Werror` and for this the unused variables would need to be uncommented:
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/aac251eeb9bc4ce1f60008d2f124863c33071e4a/src/SparkFun_Ublox_Arduino_Library.cpp#L3849
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/aac251eeb9bc4ce1f60008d2f124863c33071e4a/src/SparkFun_Ublox_Arduino_Library.cpp#L4004
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/aac251eeb9bc4ce1f60008d2f124863c33071e4a/src/SparkFun_Ublox_Arduino_Library.cpp#L4007
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/aac251eeb9bc4ce1f60008d2f124863c33071e4a/src/SparkFun_Ublox_Arduino_Library.cpp#L4008
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/aac251eeb9bc4ce1f60008d2f124863c33071e4a/src/SparkFun_Ublox_Arduino_Library.cpp#L4009

closed 2020-12-05 by PaulZC