sparkfun/SparkFun_Ublox_Arduino_Library #149: Value never used

rubienr, 2020-11-16:
Vale is never used:
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/97bd455b0e7a05b92bfe5a528c28cc47f2e2d289/src/SparkFun_Ublox_Arduino_Library.cpp#L2313

Suggested fix:
```
  //payloadCfg is now loaded with current bytes. Get what we need
  uint16_t measurementRate {extractInt(0)}; //Pull from payloadCfg at measRate LSB

  measurementRate = 1000 / measurementRate; //This may return an int when it's a float, but I'd rather not return 4 bytes
  return (measurementRate);
}
```

PaulZC, 2020-11-17:
Hi @rubienr ,
I think the original code is OK? It creates measurementRate, sets it to zero and then loads it with the 0'th byte of payloadCfg using extractInt.
Please check.
Best wishes,
Paul

rubienr, 2020-11-17:
The code is okay. Clang claims that the initialization value `0` is never used but then is overwritten in L 2316 with assignment of  `extractInt(0)`. The `measurementRate` initialization could go from L 2313 to L 2316  as `uint16_t measurementRate {extractInt(0)};` and clang would be happy again.

PaulZC, 2020-11-18:
Hi @rubienr ,
OK. However I fear that life is really much too short to make every piece of code 100% clang-compatible! Especially retrospectively...
Let me know if [this commit](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/commit/5ecba8998aa6cf2825dffe7b1adad8c1821112f4) fixes it.
Best wishes,
Paul

closed 2020-12-05 by PaulZC