sparkfun/SparkFun_Ublox_Arduino_Library #151: Conflict with IridiumSBD I2C library and SerialUSB (Qwiic Micro)

adamgarbo, 2020-11-22:
Hi there,

I've encountered an issue when attempting to use a SparkFun Qwiic Micro with a RockBLOCK 9603 and the IridiumSBD I2C library. The issue is described in detail here: https://github.com/sparkfun/SparkFun_IridiumSBD_I2C_Arduino_Library/issues/2

To summarize, the u-blox library is defining all instances of `Serial` as `SerialUSB`. However, the Qwiic Micro (one of the SerialUSB boards) uses `Serial` to connect to the RockBLOCK but this is being overwritten by the u-blox library as `SerialUSB`.

Commenting out line 63 resolves the issue:
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/97bd455b0e7a05b92bfe5a528c28cc47f2e2d289/src/SparkFun_Ublox_Arduino_Library.h#L63

Cheers,
Adam

PaulZC, 2020-11-23:
Hi Adam (@adamgarbo),
Thanks for finding this!
I suspect that if we change [this line](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/master/src/SparkFun_Ublox_Arduino_Library.h#L604), the issue will go away...
I'll run some tests.
Cheers,
Paul

PaulZC, 2020-11-23:
Hi Adam,
I'm not completely happy with this proposed fix - it is inelegant but seems to do the trick. Can you please test it?
https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/commit/75eedde568768ea967933cad40373dc9ec426f61
Many thanks,
Paul

adamgarbo, 2020-11-23:
Hi @PaulZC,

The fix appears to resolve the issue. In the case of the Qwiic Micro, it looks like SerialUSB will either need to be used explicitly, or a serial port definition other than `Serial` will need to be used (e.g. `#define SERIAL_PORT SerialUSB`).

I'd be curious to learn more about how the Arduino compiler and how it handles conflicting #define statements. If it was done in order, the two define statements below should in theory work. I don't think it's that simple, though!

```
#define Serial SerialUSB
#define IridiumSerial Serial
```

Cheers,
Adam

closed 2020-12-05 by PaulZC