sparkfun/SparkFun_Ublox_Arduino_Library #158: Feature Addition : Extend this library for Linux ARM SBCs

balamuruganky, 2020-12-10:
### Subject of the issue
===============
Since this library written in C, why should it should be restricted to Arduino only. I am proposing solution to extend this library to Linux boards like Raspberry PI, Jetson, Beaglebone etc.,

Please have a look at this [repository](https://github.com/balamuruganky/ublox_parser). This is just an experimental and the below changes needed to expand the library.

1) Implement serial Arduino library equivalent.
2) Implement serial debug Arduino library equivalent.
2) Implement print Arduino library equivalent.
3) Implement i2c Arduino library equivalent.

Serial and Serial debug libraries are implemented and print library ported partially. I2C library yet to developed.

Print functionality porting would impact the code base more than expected. Shall I go ahead and implement, Would you open to adopt once it is tested in Linux? @nseidle @PaulZC Pleasae share your thoughts.

Thanks,

nseidle, 2020-12-10:
We would be thrilled to increase the applicability of this library. We welcome the collaboration and PRs. Currently we only have resources to test and support the Arduino ecosystem but if you can handle the testing and documentation for additional platforms, we welcome it.

If you're serious, I would recommend creating very small, incremental PRs that allow us to move from Arduino dedicated to more generic platform support. For example, and perhaps you have better ideas, but a ublox_parser.cpp with hardware specific headers for other platforms that implement the serial and I2C interfaces. 

The I2C interface for u-blox is not trivial and has been heavily tested on this library so we'll be hesitant to make any large changes there. But assuming you can wall off the I2C specific interfaces to specific platforms then I think we can make it work.

We'll always want and need to maintain [Arduino Library compatibility](https://arduino.github.io/arduino-cli/library-specification/) as that is our current core user base.

balamuruganky, 2020-12-20:
@nseidle @PaulZC : Please have a look at the https://github.com/balamuruganky/SparkFun_Ublox_Arduino_Library repository, "linux_platform" branch. 

"linux" folder is been added to port the Arduino APIs to Linux and the header file changed like below,

#ifdef LINUX_PLATFORM

#include "Stream.h"
#include "Utils.h"
#include "Wire.h"

#else

#if (ARDUINO >= 100)
#include "Arduino.h"
#else
#include "WProgram.h"
#endif

#include <Wire.h>

#endif

It is a proof of concept code base. Serial communication and Print functionality is beed tested. I2C yet to be tested. 
Please share your comments, if you have any.

Thanks,
Bala

balamuruganky, 2021-01-14:
@PaulZC, @nseidle : I have created separate repository for Linux porting and used this code base as submodule. Please have a look at https://github.com/balamuruganky/Ublox_Linux_Library. Feel free to refer https://github.com/balamuruganky/Ublox_Linux_Library for C++ Linux library. Thanks.

balamuruganky, 2021-01-14:
Closing this issue as it is concluded.

closed 2021-01-14 by balamuruganky