sparkfun/SparkFun_Ublox_Arduino_Library #165: sendCommand (poll / get) failed!  for UBX_RXM_RAWX request

skywo1f, 2020-12-24:
set up the message:
`  customCfg.cls = UBX_CLASS_CFG; // This is the message Class
  customCfg.id = UBX_RXM_RAWX;          //UBX-RXM-RAWX
  customCfg.len = 0; // Setting the len (length) to zero let's us poll the current settings
  customCfg.startingSpot = 0; // Always set the startingSpot to zero (unless you really know what you are doing)
`

then sent with 
`if (myGPS.sendCommand(&customCfg, maxWait) != SFE_UBLOX_STATUS_DATA_RECEIVED) `

but that seems to have failed. This is on the sam-m8q breakout board. I am talking to it via arduino

PaulZC, 2020-12-24:
Hi @skywo1f ,
Tricky... Accessing RAWX via customCfg _should_ work but:
- RAWX messages can be around 2KBytes in size. You will need to make sure your "payload" is big enough to hold the entire message.
- RAWX works best when sent "periodically" (via CFG MSG) - the library doesn't support that - yet...
- And there may be a gotcha in the library that prevents messages larger than MAX_PAYLOAD_SIZE from being received...
- The final gotcha is that RAWX is only available on "High Precision" or "Time Sync" products like the ZED-F9P. It won't work on the SAM-M8Q.

The good news is that RAWX will be supported in the new v2.0 of the library. We are trying to have this ready by mid-January.
Best wishes,
Paul
[image]

PaulZC, 2021-01-15:
v2.0 of the library is here:
https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library
And the data logging examples are here:
https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/tree/master/examples/Data_Logging

But please remember that you cannot receive RAWX on the SAM-M8Q. You need a high precision module like the ZED-F9P.

skywo1f, 2021-01-18:
Looking at the datasheet for the SAM-M8Q:
https://www.u-blox.com/sites/default/files/SAM-M8Q_ProductSummary_%28UBX-15027452%29.pdf
it states:
Raw Data: Code phase output

Is that not the data required to find satellite range?

PaulZC, 2021-01-18:
Hi @skywo1f ,

OK. I agree. "Raw Data: Code phase output" certainly suggests you should be able to access the carrier phase data (via RAWX)...

But if you look at the M8 protocol specification, there is a note that RAWX is only available on certain modules (not including the SAM-M8Q as that is a "standard precision" product)

[image]

I think I'd better test this! It would be a massive plus if you can receive RAWX on the SAM. Certainly it wasn't true a few years ago. I had to buy a NEO-M8T to be able to get RAWX back then...

Best wishes,
Paul

skywo1f, 2021-01-18:
Thanks!
Is that command supported? do you know if the process is the same as rawx?

PaulZC, 2021-01-18:
RXM-MEASX is not (currently) one of the 23 "automatic" messages supported by v2.0 of the library.
(Here's the current list: https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/Theory.md#auto-messages)
We could add it, but I'm not sure if it would give you the information you're looking for... I suspect you probably do need RAWX?
And, sadly, we don't have a SparkFun Qwiic NEO-M8T breakout... The NEO-M8T is _probably_ still your cheapest way of getting RAWX data.

skywo1f, 2021-01-18:
I agree RAWX would be ideal, but I should at least be able to get 3-6 meter accuracy  (hopefully not 300m) from the code phase (Distance to satellite) . With a few satellites it might add up to something useful. 

If you ever get around to supporting MEASX, please let me now. Thanks!

PaulZC, 2021-01-22:
Hi @skywo1f ,
I'm closing this issue as we are about to deprecate this version of the library. Please feel free to open a "Feature Request" issue in v2.0 of the library if you would like us to add RXM-MEASX support. It is really not clear to me how much improvement this would give you, but we can certainly consider adding support for it.
Best wishes,
Paul

closed 2021-01-22 by PaulZC