sparkfun/SparkFun_Ublox_Arduino_Library #166: Update example library links

adamgarbo, 2020-12-30:
Hey folks,

Good to see the renamed u-blox library was updated on the Arduino IDE. I didn't realize the name change would require an entirely new library to be downloaded. This may throw other current users for a loop as it's not an automatic update.

A quick heads up that the name change has broken all links to the library in the examples:

`//http://librarymanager/All#SparkFun_Ublox_GPS`

It looks like the link will need to be changed to either of the following:

1) `//http://librarymanager/All#SparkFun_u_blox_GPS`
2) `//http://librarymanager/All#SparkFun_u-blox_GPS`

Cheers,
Adam

PaulZC, 2020-12-30:
Thanks Adam,
We're going to update the library helper links in v2.0. Possibly 1.8 too - if time permits.
Best,
Paul

PaulZC, 2020-12-30:
Thanks for the help Adam!
Closing...

closed 2020-12-30 by PaulZC