sparkfun/SparkFun_Ublox_Arduino_Library #174: myGPS.hnrPVT.nano outputting -1 all the time?

bvernham, 2021-01-06:
### Subject of the issue
Describe your issue here. If you reference a datasheet please specify which one and in which section (ie, the protocol manual, section 5.1.2). Additionally, screenshots are easy to paste into github.

### Your workbench
* What development board or microcontroller are you using?
* due
* What version of hardware or breakout board are you using? 
* sparkfun m8u
* How is the breakout board wired to your microcontroller?
* Serial3
* How is everything being powered?
* Due on board 5v ps
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue.
Serial<<myGPS.hnrPVT.nano<<"\n";
hnrPVT always returns -1 for nano
### Expected behavior
Tell us what should happen
nano should have a real varying value
### Actual behavior
Tell us what happens instead
hnrPVT always returns -1 for nano

PaulZC, 2021-01-07:
Hi Bruce (@bvernham),

[Looking at the code](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/master/src/SparkFun_Ublox_Arduino_Library.cpp#L1239), nano is being extracted correctly. So I don't see a problem there.

Also, -1 is a valid value for nano. nano is actually an offset in the range -1e9 .. 1e9 nanoseconds (ie. +/- 1 second). It is a ```signed long``` (int32_t). You need to _**add**_ it to iTOW to get the time now with nanosecond resolution - and you'll need 64-bit math to do that. (I'm not sure if the DUE implements 64-bit signed integers?)

[image]

All the best,
Paul

closed 2021-01-07 by PaulZC

bvernham, 2021-01-07:
Ok that makes some sense.  So with the hnr messages there is no easy way to get millisecond like the nav-pvt?

____________________________________________________________Top News - Sponsored By NewserRelatives Identify Woman Fatally Shot Inside CapitolTrump: 'There Will Be an Orderly Transition'Congress Certifies Biden's Win

PaulZC, 2021-01-07:
iTOW is in ms. Try ```uint16_t millisecs = hnrPVT.iTOW % 1000;```

bvernham, 2021-01-07:
Thanks,,,On another subject I seems to be struggling with the serial communication now. I am running the the auto HNR message @ 10 hz on serial3 and sometimes hnrATT will pop up and true multiple times for each time PVR and DYN become true? Buffer overrun?  Communication issue?  How would I know if there is a checksum error? I know it is off subject but I have been thinking of opening another issue but if there is a simple resolution I would like to avoid another "issue". Thanks

 

&mdash;
Sponsored by https://www.newser.com/?utm_source=part&utm_medium=uol&utm_campaign=rss_taglines_more

http://thirdpartyoffers.juno.com/TGL3131/5ff70d8c91afdd8c3036st04vuc1
How Was Capitol Breach Possible? 'I Can't Figure That Out'
http://thirdpartyoffers.juno.com/TGL3131/5ff70d8cb0d9bd8c3036st04vuc2
Late Night's Take on Wednesday Was Pretty Bleak
http://thirdpartyoffers.juno.com/TGL3131/5ff70d8cd02d3d8c3036st04vuc3

PaulZC, 2021-01-07:
Hey Bruce,
Buffer overrun. You need to make sure you are calling ```checkUblox``` fast enough to prevent the data overrunning the serial receive buffer. v2.0 will make this easier for you - and I'm almost done there... Almost...
Paul

bvernham, 2021-01-07:
I used Serial3.availible()>3 every loop but it seems make things worse?I am using millis timer which is 1/16th of the update rate which seems excessive (100 ms/16) which I woud think it would be enough. Due buffer size is 64 bytes but can be increase by tweaking the core files. The HNR messages are 72 + 36 + 32 is 140 bytes so go with 256 bytes? Any thoughts would be helpful. Thanks Bruce

 

http://thirdpartyoffers.juno.com/TGL3131/5ff71f777dcb51f777092st01vuc1
http://thirdpartyoffers.juno.com/TGL3131/5ff71f77a175f1f777092st01vuc2
http://thirdpartyoffers.juno.com/TGL3131/5ff71f77c52b81f777092st01vuc3

PaulZC, 2021-01-07:
Time to reach for the calculator... 140 bytes is 1400 bits (at 8+1+1 bits/char). Divide by your Baud rate to find out how long the data takes to arrive. Your buffer needs to be large enough and you need to call checkUblox often enough to stop the buffer overflowing.

bvernham, 2021-01-07:
1400/115200 = 0.01215277777777777777777777777778 I am checking every 6 ms. Why wouldn't Serial3.availible()>0 do the trick? Only call get when there is data in the buffer and do that every loop. I looked in checkublox and that is all it does is check the same thing, right? Thanks Bruce 

 

http://thirdpartyoffers.juno.com/TGL3131/5ff726649f482663589fst01vuc1
http://thirdpartyoffers.juno.com/TGL3131/5ff726642d4422663589fst01vuc2
http://thirdpartyoffers.juno.com/TGL3131/5ff726645149e2663589fst01vuc3

PaulZC, 2021-01-07:
If your serial buffer is 64 bytes, at 115200 Baud it will fill up in 5.6ms. Try 3ms to provide some margin?

bvernham, 2021-01-08:
Serial buffer was originally 128 I increased to 512 and move the get statements into serialeventrun() which seems to help.  I can get up to about 25 hz before I start seeing the repeats.  Maybe now I need to increase the UART baud rate again. iTOW was exactly what I was looking for and work as exacted.
I just wish I could find a good way to confirm the due memory usage after the compile is complete. Thanks for all help. ---------- Original Message ----------

 

Sponsored by https://www.newser.com/?utm_source=part&utm_medium=uol&utm_campaign=rss_taglines_more

Controversial Senator Loses His Book Deal
http://thirdpartyoffers.juno.com/TGL3131/5ff84d011b78a4d00471dst04vuc1
'Hitler Was Right on One Thing,' Lawmaker Says
http://thirdpartyoffers.juno.com/TGL3131/5ff84d014003b4d00471dst04vuc2
Capitol Police Officer Dies After Wednesday's Riot
http://thirdpartyoffers.juno.com/TGL3131/5ff84d01637b74d00471dst04vuc3