sparkfun/SparkFun_Ublox_Arduino_Library #176: Hardware connection section has information for the serial reset sketch but isn't needed for I2C

cmfrancis, 2021-01-14:
Hardware connection has information for the serial reset sketch but isn't needed for I2C

PaulZC, 2021-01-15:
Thanks Chris,
Fixed in [this commit](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/commit/6595144038a8573b30f7c28aee245e37fa83f686). It is currently still in the release_candidate branch because:
We are about to deprecate the "SparkFun Ublox Arduino Library" and are going to encourage users to migrate over to the shiny new v2.0 "[SparkFun u-blox GNSS Arduino Library](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library)".
I've updated the reset example there too.
We'll update the example here and do the deprecation all in one go. I'll leave this open until we do.
Cheers!

PaulZC, 2021-01-22:
This will be fixed in v1.8.11 - which will go live shortly.

closed 2021-01-22 by PaulZC