sparkfun/SparkFun_Ublox_Arduino_Library #177: Add  auto reporting for UBX-ESF

bvernham, 2021-01-17:
### Subject of the issue
Describe your issue here. If you reference a datasheet please specify which one and in which section (ie, the protocol manual, section 5.1.2). Additionally, screenshots are easy to paste into github.
Add  auto reporting for UBX-ESF
### Your workbench
* What development board or microcontroller are you using?
* What version of hardware or breakout board are you using? 
* How is the breakout board wired to your microcontroller?
* How is everything being powered?
* Are there any additional details that may help us help you?

### Steps to reproduce
Tell us how to reproduce this issue. Please post stripped down example code demonstrating your issue.

### Expected behavior
Tell us what should happen
UBX-ESF dynamics is gravity free but the HNR messages include gravity.

Still have not got an answer of what the difference is meant between "Compensated x-axis acceleration (gravityfree). and Compensated x-axis acceleration (with gravity) on either the UBLOX or the SparkFun Forum.

### Actual behavior
Tell us what happens instead

PaulZC, 2021-01-18:
Hi Bruce (@bvernham ),

"Auto" ESF is fully supported in the new v2.0 of the library. You can find it here:
https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library

The new callback examples are here (there are examples for ESF and HNR):
https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/tree/master/examples/Callbacks

I interpret "gravity free" as the sensor values _excludes_ gravity. I.e. you will _not_ see a static 9.8 m/s^2 component.
I interpret "with gravity" as the sensor values _include_ gravity. I.e. you _will_ see a static 9.8m/s^2 on whichever axis is pointing 'down'.

Best wishes,
Paul

closed 2021-01-18 by PaulZC

bvernham, 2021-01-18:
Do you know what kind of update rates auto ESF can run at? HNR messages specifically say 30 hertz. Thanks for the great work.

 

 https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library
 https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/tree/master/examples/Callbacks
&mdash;
Sponsored by https://www.newser.com/?utm_source=part&utm_medium=uol&utm_campaign=rss_taglines_more

Among Charges in Capitol Riot, a 'Bizarre Claim' About Russia
http://thirdpartyoffers.juno.com/TGL3131/6005879317524793221cst04vuc1
Man Found Living at O'Hare for Months Cited COVID Fears
http://thirdpartyoffers.juno.com/TGL3131/6005879339c73793221cst04vuc2
Trump Has Busy Last Full Day in Office Planned
http://thirdpartyoffers.juno.com/TGL3131/600587935d1d9793221cst04vuc3

PaulZC, 2021-01-18:
Check the datasheet and/or integration guide for your module

bvernham, 2021-01-20:
Can you disable the callbacks for V2 library?

Thanks

PaulZC, 2021-01-20:
Yes, just don't use them. The callback is only enabled if you call (e.g.) myGNSS.setAutoESFALGcallback
You can still use polling: call (e.g.) myGNSS.getEsfAlignment
And you can still use "auto" (without callback): call (e.g.) myGNSS.setAutoESFALG
Access the data by reading (e.g.) myGNSS.packetUBXESFALG->data.yaw
Mark the data as stale by calling (e.g.) myGNSS.flushESFALG
Have a look at the [Dead Reckoning examples](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/tree/master/examples/Dead_Reckoning) 5 & 6